edk2-porting/edk2-msm #11: Fix DeviceMemoryMap.h

strongtz, 2021-03-17:
The current memory map is optimized for LA use cases, which is not Windows-friendly.
In order to have working GPU, WiFi, Modem etc. in Windows, we must reserve the related PIL (Peripheral Image Loader) memory regions during edk2 init.

strongtz, 2021-03-19:
We only need to fix the Windows drivers

strongtz, 2021-06-04:
probably there's still something broken

xqymain, 2021-07-31:
I tried to modify 

`gArmTokenSpaceGuid.PcdSystemMemorySize|0x140000000`

 but it can't work. I only saw this issue now. 

There is currently no way to expand more available memory, right?

Idonotkno, 2021-08-01:
@xqymain It's no use just changing this. At the same time, why do you want to change it?

xqymain, 2021-08-01:
@Idonotkno Well, the available memory in Windows is only 5.3G. And I want to get more memory to run the program.

Idonotkno, 2021-08-01:
@xqymain Modify the memory size in memorymap and dsc, but it should not exceed 7.5G, and it should be increased at the same time. If the difference between uefi set memory and mapped memory is too large, the initialization of mmu will fail

xqymain, 2021-08-01:
@Idonotkno I don't know how to modify in MemoryMap. I have read DeviceMemoryMap.h carefully, but I still don't know which part I should modify.

xqymain, 2021-08-02:
I have tried to change a lot of things, but have been unsuccessful. I only changed dsc and DeviceMemoryMap.h now, and now it runs on 7.5G memory.

> 
> 
> @Idonotkno I don't know how to modify in MemoryMap. I have read DeviceMemoryMap.h carefully, but I still don't know which part I should modify.

strongtz, 2021-09-21:
It has been fixed.

closed 2021-09-21 by strongtz