edk2-porting/edk2-msm #15: Windows getting activated automatically

strongtz, 2021-05-19:
Currently, Windows booted using edk2-sdm845 will be activated automatically once connected to Internet. It might be due to someone activating using Digital Rights.

This may lead to unexpected legal risks.

To solve the issue, generating a unique SMBIOS UUID at runtime based on device serial number might be needed. EFIChipInfo protocol might be required.

It is currently under investigation by @BigfootACA

Vicc2008, 2021-05-21:
Hello i have a Motorola One Action and I want Port edk2 UEFI to my Motorola One Action

Vicc2008, 2021-05-21:
Hello i have a Samsung Galaxy A30 and I want Port edk2 UEFI to my Samsung Galaxy A30

ivoszbg, 2021-05-24:
This is the same case as windows 10 on armv7 lumias - by default devices under 9 inches get automatically activated (https://www.theverge.com/2014/4/2/5574146/microsoft-making-windows-free-on-devices-with-screens-under-nine)

closed 2021-06-24 by strongtz