edk2-porting/edk2-msm #25: Poco F1?

RusJJ, 2021-06-14:
Any plans on fixing touch and GPU for Poco F1? Atleast touch, please?

TAOcroatia, 2021-06-14:
Touchscreen is currently not our priority, but we will start working on it eventually. If you have a focaltech model, you can try out a driver on edk2-porting but you will need to build it yourself. As for the gpu, we are working on it but don't have expectations of it working any time soon. 

If you want to speed the process up and know some developers familiar with writing windows drivers or messing with edk2, point them to this project😄

closed 2021-06-24 by strongtz

RusJJ, 2021-07-27:
Hey, just wanna ask. Is touchscreen working now? Or it's still in plans? Because i seen your branch

TheMorc, 2021-07-27:
Still nothing yet but I think it's still planned.