edk2-porting/edk2-msm #34: Google Pixel 3/3XL support

nickobelic1999, 2021-07-06:
### Prerequisites

* [ X] Have you read the readme?
* [ X] Do you (want to) have your device in the supported list?
* [ X] Does the device have a Snapdragon 845 SOC?

### Description

Hey I saw the readme which said to not use a Google Pixel 3/3XL as it will wipe the UFS clean. Is there any chance that this could work on these devices since they are Snapdragon 845 devices?

I own a Pixel 3XL and would be happy to test / help (provided it won't brick the device :yum: )

TheMorc, 2021-07-06:
Yes, it will work on them but it'll wipe the UFS.
And that's something you really don't want to happen.
That's why they aren't supported even if they're SDM845.

strongtz, 2021-07-08:
We already have several bricked Pixel 3 :)

closed 2021-07-08 by strongtz

leonvking0, 2023-07-10:
Luckily I read this before trying anything on my pixel 3XL lol..
Will it be supported though?

leonvking0, 2023-07-10:
> We already have several bricked Pixel 3 :)

> We already have several bricked Pixel 3 :)

Luckily I read this before trying anything on my pixel 3XL lol..
Will it be supported though?