edk2-porting/edk2-msm #36: Just a question: What is the UFS in google and sony phones?

cuj, 2021-07-10:
Hi, I have read the README.MD file, and I read the following:

> # WARNING
> DO NOT EVER TRY TO PORT IT TO SONY and GOOGLE DEVICES
> YOUR UFS WILL BE WIPED CLEAN!!!

I do not understand the meaning of UFS, so is UFS the system files/data of sony and google devices? is it responsible for booting devices? or is UFS the system files of an additional feature only on google/sony phones? it would be much appreciated if someone can tell me what's the UFS. thanks.

strongtz, 2021-07-10:
@cuj Google "UFS" pls

cuj, 2021-07-10:
@strongtz

I googled, and the UFS is like the SSD of the sony/google phone, just to be sure, is that true? Thanks.

NXij, 2021-07-10:
UFS is the storage chip, mobile SOCs keep everything on them, hence when it gets wiped, it essentially bricks the phone with no ability to boot into any kind of uefi/bootloader/fastboot (which includes the ability of being able to flash anything at all).

There is a basic SDL2 port for Pixel 3XL which had its last commit in 2019 though, it does not seem to wipe the ufs, but afaik it does also not have any way to interface with the on device UFS.

cuj, 2021-07-11:
@NXij

Thanks for telling me what is UFS.

cuj, 2021-07-12:
Reopening the issue as I have another question:

If I try to port win10 edk2 to sony or google phone, Will the UFS instantly get wiped, or will Windows 10 boot but after a reboot the device is bricked?

strongtz, 2021-07-31:
It will get wiped instantly

closed 2021-07-31 by strongtz

Bezouro, 2024-07-15:
@cuj please dont edit your messages, it make dificult to ppl like me to clear the same doubts you had