edk2-porting/edk2-msm #42: can you make a port for poco x3 pro

kwteen, 2021-07-22:
### Prerequisites

* [ ] Have you read the readme?
* [ ] Do you (want to) have your device in the supported list?
* [ ] Does the device have a Snapdragon 845 SOC?

### Description

[Description of the bug or feature]

**Expected behavior:** [What you expected to happen]

**Actual behavior:** [What actually happened]

TheMorc, 2021-07-22:
[image]
SDM860 is a tweaked version of 855 and ***this is a SDM845 repo***.
The 855 repo is not available right now on GitHub.

strongtz, 2021-07-22:
@kwteen Does the device have a Snapdragon 845 SOC?

BigfootACA, 2021-07-22:
this repo only accepts sdm845

closed 2021-07-22 by BigfootACA

proganime1200, 2022-05-05:
There is a working uefi kernel for x3 pro on 4pda https://4pda.to/forum/index.php?showtopic=884586&st=2340