edk2-porting/edk2-msm #45: Question:Support for "OPPO Reno A".

creeper-0910, 2021-07-29:
"OPPO Reno A" is equipped with Snapdragon 845, but it is not supported.
Do you plan to support it?

TheMorc, 2021-07-29:
We don't have the devices so if you're willing to, you can port it

Idonotkno, 2021-07-29:
@creeper-0910 Findx and oneplus have been tested, and other OPPO devices can work in theory, but you need to test your own functions

creeper-0910, 2021-07-29:
> 私たちはデバイスを持っていないので、あなたが喜んでそれを移植することができます
> 
> https://forum.renegade-project.org/t/edk2-sdm845-new-device-porting-guide/417

Do I need root privileges on my device to run this?

BigfootACA, 2021-08-02:
> > 私たちはデバイスを持っていないので、あなたが喜んでそれを移植することができます
> > https://forum.renegade-project.org/t/edk2-sdm845-new-device-porting-guide/417
> 
> Do I need root privileges on my device to run this?

the bootloader must be unlocked

creeper-0910, 2021-08-05:
I'm sorry.
It seems that SDM710 was displayed as 845 due to a bug(?) in CPU-Z...
Oppo reno A is not SDM845...

closed 2021-08-05 by creeper-0910