edk2-porting/edk2-msm #47: Enable 3D acceleration support

10-p, 2021-08-03:
### Prerequisites

* [✓] Have you read the readme?
* [✓] Do you (want to) have your device in the supported list? OnePlus 6
* [✓] Does the device have a Snapdragon 845 SOC? Yes

### Description

Enable GPU support.

[WOA drivers](https://github.com/edk2-porting/WOA-Drivers/issues/4) need support from here to be able to enable 3D acceleration.

**Expected behavior:** Complete Windows installation yields 3D acceleration as seen in tutorials and other sources or videos.

**Actual behavior:** Everything worked but 3D acceleration is unusable or just not available.

TheMorc, 2021-08-04:
@10-p Renegade doesn't support GPU yet, adding the drivers to WOA-Drivers would be a useless waste of time

TheMorc, 2021-08-04:
@10-p Just relax, it'll come one day

TheMorc, 2021-08-21:
> @10-p Just relax, it'll come one day

Aaand it came! Only for Mi Mix 2S for now but support for other devices will eventually come later.

sssolitude, 2021-08-22:
Excellent! so cool! maybe, could the dipper(mi 8) get the GPU support the next one! can`t wait anymore!

sssolitude, 2021-08-22:
### Prerequisites

* [√ ] Have you read the readme?
* [√] Do you (want to) have your device in the supported list?
* [ √] Does the device have a Snapdragon 845 SOC?

### Description

windows11 ran successfully on my mi8 with the release driver and img this morning ,and about 17:00 I saw the GPU update. So and download the img in Action CI#177 and drivers the latest. I use fastboot boot the newest img and upgrade the drivers using dism++ directly in windows11, then reboot also with the img. but the GPU status is as follows:

Device status:
Windows has stopped this device because it has reported problems. (Code 43)

Device ACPI\VEN_QCOM&DEV_027E&SUBSYS_RENEGA0E&REV_007C\0 had a problem starting.

Driver Name: oem50.inf
Class Guid: {4d36e968-e325-11ce-bfc1-08002be10318}
Service: QCDX
Lower Filters:
Upper Filters:
Problem: 0x26
Problem Status: 0xC000038E

**Expected behavior:** [What you expected to happen]

**Actual behavior:** [What actually happened]

BigfootACA, 2021-08-23:
now supported

closed 2021-08-23 by BigfootACA