edk2-porting/edk2-msm #48: 一加6 锁屏后触摸屏 可能会失灵，触摸长按没有右键菜单

niuniu132, 2021-08-07:
### Prerequisites

* [ ] Have you read the readme?
* [ ] Do you (want to) have your device in the supported list?
* [ ] Does the device have a Snapdragon 845 SOC?

### Description

[Description of the bug or feature]

**Expected behavior:** [What you expected to happen]

**Actual behavior:** [What actually happened]

strongtz, 2021-08-07:
@niuniu132 That's expected.
Also, please use English.

cruisermaxx, 2021-08-07:
I also find the touch screen stop problem. Could this be fixed on oneplus 6/6t?

TheMorc, 2021-08-07:
We know about the current issues
https://forum.renegade-project.org/t/current-synaptics-touch-driver-issues/204

TheMorc, 2021-08-24:
> I also find the touch screen stop problem. Could this be fixed on oneplus 6/6t?

The "touchscreen stop" problem has been fixed for now. It shouldn't happen now with the latest drivers but you'll still have issues with right clicking.

TheMorc, 2021-09-08:
@cruisermaxx @niuniu132

Try the latest driver from WOA-Drivers, it should work now on your replaced screen with a counterfeit digitizer controller IC.

TAOcroatia, 2021-09-08:
Issue already open on different repo

closed 2021-09-08 by TAOcroatia