edk2-porting/edk2-msm #51: Poco f1 window sinstall Tutorial please

hwanghyunseo, 2021-08-22:
There is no way to install Windows arm os on poco f1 on YouTube.  There is only a video in Portuguese, but it is an introduction video, not a method.  Please tell me how to install on pocophone f1.

hwanghyunseo, 2021-08-22:
Portuguese video link:https://youtu.be/t81-Ji17fGM

salam-siddique, 2021-08-24:
installed successfully on 2 poco f1 phones

hwanghyunseo, 2021-08-24:
> installed successfully on 2 poco f1 phones
How to?

Tonio1987fr, 2021-09-07:
The guide for the oneplus 6 works. You just have to modify the partitions for the Pocophone.

Attention the touchscreen does not work on the Pocophone f1 =(

closed 2021-09-08 by hwanghyunseo

neozeed, 2023-06-29:
> installed successfully on 2 poco f1 phones

What version of Windows with which bootloader?
Release - 2301.1 for the ebbg doesn't boot windows at all, but it seems older ones do?  But I always crash with MEMORY_PROTECTION or IRQL less than expected.

ComitC3, 2023-06-29:
Hey, it's because of some driver issues(maybe) I have used some old drivers and removed one. And now all the thing works(that's supposed to work). Touch also works. I also get to use that error but now everything works as stated. For fix Drivers- use usb drivers for rc02, not latest, and then if your device is FTS panel, remove the nt36xx folder of driver and only keep focaltouch folder.total there will be 61 drivers instead of 62. Also ebbg devices might have to remove pmic folder also for some devices.use the given tutorial in the installation page , its enough.

jukfiuune, 2023-06-29:
There is literally a tutorial on the website. USB and touchscreen work on my device. Only thing that I know doesn't work is the latest Windows 11 Canary and Dev build. Here: https://renegade-project.tech/en/devices/xiaomi/beryllium

neozeed, 2023-06-29:
I used 10.0.21390.1.co_release.210521-1658_arm64fre for the windows install. useing the [following link](https://uupdump.net/download.php?id=6db21a1e-53af-48cb-a0ba-94e1b852c33b&pack=en-us&edition=professional) to make sure it got the updates etc.

I assume the drivers in question is [v2.0rc2](https://github.com/edk2-porting/WOA-Drivers/releases/download/v2.0rc2/beryllium.tar.gz)

I added the drivers, made sure to Enable test signing and disable Automatic Repair

I used the bootloader: boot-beryllium-ebbg Release - 2210.1.img as 23xx doesnt boot anything.

and blinking like crazy 
[image]

ComitC3, 2023-06-29:
I think I should try the way I stated. It might work. Worked for me at least I too couldn't boot in latest 2301 img. If ebbg driver, u need to try removing the pmic driver folder before installing drivers

neozeed, 2023-06-29:
[image]

So here is how I got here...

Prep the disk, format etc etc.. nothing too special?
```
C:\Users\neozeed\Downloads\adb>diskpart

Microsoft DiskPart version 10.0.19041.964

Copyright (C) Microsoft Corporation.
On computer: ERYING

DISKPART> list disk

  Disk ###  Status         Size     Free     Dyn  Gpt
  --------  -------------  -------  -------  ---  ---
  Disk 0    Online          953 GB  1024 KB        *
  Disk 1    Online          953 GB  1024 KB        *
  Disk 2    Online         1863 GB      0 B
  Disk 3    Offline          32 MB  8128 KB        *
  Disk 4    Offline          32 MB    28 MB        *
  Disk 5    Offline           8 MB   384 KB        *
  Disk 6    Offline        4608 MB    63 MB        *
  Disk 7    Offline           8 MB   384 KB        *
  Disk 8    Offline         114 GB      0 B        *

DISKPART> select disk 8

Disk 8 is now the selected disk.

DISKPART> online

Microsoft DiskPart version 10.0.19041.964

DISK        - Online a disk that is currently marked as offline.
VOLUME      - Online a volume that is currently marked as offline.

DISKPART> online disk

DiskPart successfully onlined the selected disk.

DISKPART> list volume

  Volume ###  Ltr  Label        Fs     Type        Size     Status     Info
  ----------  ---  -----------  -----  ----------  -------  ---------  --------
  Volume 0     C                NTFS   Partition    953 GB  Healthy    Boot
  Volume 1                      FAT32  Partition    100 MB  Healthy    System
  Volume 2                      NTFS   Partition    530 MB  Healthy    Hidden
  Volume 3     D   Data         NTFS   Partition    953 GB  Healthy
  Volume 4     U   TOSHIBA EXT  NTFS   Partition   1863 GB  Healthy
  Volume 5         SYSTEM       FAT32  Partition    476 MB  Healthy    Hidden
  Volume 6         Windows      NTFS   Partition     84 GB  Healthy

DISKPART> select volume 5

Volume 5 is the selected volume.

DISKPART> format quick fs=fat32 label="System"

  100 percent completed

DiskPart successfully formatted the volume.

DISKPART> assign letter="S"

DiskPart successfully assigned the drive letter or mount point.

DISKPART> select volume 6

[... 154 lines cut ...]

DISKPART> remove letter=s

DiskPart successfully removed the drive letter or mount point.

DISKPART> select volume 5

Volume 5 is the selected volume.

DISKPART> remove letter=w

DiskPart successfully removed the drive letter or mount point.

DISKPART> offline disk

DiskPart successfully offlined the selected disk.

DISKPART> exit

Leaving DiskPart...

C:\Users\neozeed\Downloads\adb>
```

I'm pretty sure it must be the combination of what version of windows vs trying to load drivers?

neozeed, 2023-06-29:
Okay some success!

In that Brazilian video he's using 210521-1658 so its's def a known good version.  In the 2210.1-fix drivers I saw there is a minimal directory so I applied that, while swapping the USB from beryllium v2.0rc2.  I also ran the 'touchfixup', and of course all the BCD flags.  I did add this flag:

`bcdedit /store BCD /set "{default}" debug on`

Oddly enough the 2301.1 bootloader work!?  First boot it does the getting ready and reboots.  Second boot it went into the setup steps, and I got a OTG adapter, plugged in a mouse and completed the setup.

Touchscreen didn't work, and there was no networking.  But it did boot!

```
10.0.21390.1.co_release.210521-1658_arm64fre
boot-beryllium-ebbg Release - 2301.1.img	2210.1-fix\minimal	getting ready	reboot	
									no usb no touch

```