edk2-porting/edk2-msm #64: Oneplus 6T, Qualcomm Crushdamp Mode

cruisermaxx, 2021-09-30:
### Prerequisites

* [Yes] Have you read the readme?
* [Already in] Do you (want to) have your device in the supported list?
* [Yes] Does the device have a Snapdragon 845 SOC?

### Description
Oneplus 6T, trigger the "Qualcomm Crushdamp Mode" frequently every day. If the mobile go into sleep mode.
I don't know whether this situation also meet in ohter devices such as Xiaomi Mix 2s or Mix 3 and so on.

[Description of the bug or feature]

**Expected behavior:** [What you expected to happen]
Back to normal condition for daily use.

**Actual behavior:** [What actually happened]
Affect daily use seriously.

GoldRenard, 2021-09-30:
After some testing I found that this most likely caused by FG/PMIC drivers and only happens when phone is not on charger at the moment.

Not a solution but removing BattMgr driver fixes this exact issue but you're losing FG functions in this case.

cruisermaxx, 2021-09-30:
> 
> 
> After some testing I found that this most likely caused by FG/PMIC drivers and only happens when phone is not on charger at the moment.
> 
> Not a solution but removing BattMgr driver fixes this exact issue but you're losing FG functions in this case.

Great! Thank you very much for testing and providing some clues. If so I can toggle on screen off after some time when plugin the charger, and toggle off screen off function when only use battery in Win11 settings.

I will observe for some time and if I get any results I will feed back here later.
Thank you!

[image]

bladeboy2021, 2021-09-30:
This method doesn't work. I tried it a long time ago. After updating the appropriate UEFI and Win11 patches, you are now waiting for a long time to improve access to Qualcomm CrashDump Mode.

bladeboy2021, 2022-02-28:
One plus six win11 Qualcomm crash dump mode, an bug that haunts a lot of computer friends, now shares its experience of testing over and over again. By turning off dormancy and shutting down the cell phone cellular network driver (qualcomm mobile broadband device), Qualcomm crash dump mode can be basically solved, long standby to dead power will not crash, play some games suddenly darkened and never appeared again, basically can be daily. The first time you brush win11 into the system, if you don't get into the system with Qualcomm crash dump mode, you can try twrp to brush with other lower versions of UEFI, or go into pe to delete a driver that's not a core, or go into Android and switch to win11. You can go into the win11 system and you can solve it in this way.

Momoprawdziwe, 2023-01-11:
Hi I also have a problem with my op6t could you send me driver store folder?

Momoprawdziwe, 2023-01-11:
Hi I also have a problem with my op6t could you send me your driver store folder?