edk2-porting/edk2-msm #88: Clarification: UFS wipe

IngwiePhoenix, 2022-01-20:
### Prerequisites

* [x] Have you read the readme?
* [ ] Do you (want to) have your device in the supported list?
* [ ] Does the device have a Snapdragon 845 SOC?

### Description

In the README, it is stated to never port to Sony or Google devices due to the UFS being wiped. Why, exactly? I am curious and would like to know :)

**Expected behavior:**

Not a whole lot; just a little bit of clarification. I know what a UFS is - just not why it becomes wiped when trying to run this port.

**Actual behavior:**

Curiosity I guess?

Kind regards,
Ingwie

Idonotkno, 2022-01-20:
@IngwiePhoenix OTP

luka177, 2022-01-20:
@IngwiePhoenix Actually on pixel devices it is wiped not by uefi but by windows driver, so probably you can boot mainline

Idonotkno, 2022-01-20:
@luka177 It triggers the protection of ufs itself, which is the same on the main line. It can only be used if the ufs driver for sony is modified.

luka177, 2022-01-20:
@Idonotkno On Sony probably, while on pixel mainline is confirmed working with ufs, https://wiki.postmarketos.org/wiki/Google_Pixel_3(google-blueline)

IngwiePhoenix, 2022-01-20:
> @IngwiePhoenix OTP

Trying to google that leads me to a lot of "One Time Password" related apps. You likely don't mean that one, do you?

Idonotkno, 2022-01-20:
@luka177 Unfortunately, we don't have the ufs driver source code for Windows, so we can't customize it.

Idonotkno, 2022-01-20:
@IngwiePhoenix Ufs protection is turned on.

Idonotkno, 2022-01-20:
@Idonotkno See the following questions completely.

closed 2022-01-23 by BigfootACA