edk2-porting/edk2-msm #90: 求适配pixel3 （blueline）

EnderKi, 2022-01-21:
### Prerequisites

* [√ ] Have you read the readme?
* [ √ ] Do you (want to) have your device in the supported list?
* [ √ ] Does the device have a Snapdragon 845 SOC?

### Description

[Description of the bug or feature]

**Expected behavior:** [What you expected to happen]

**Actual behavior:** [What actually happened]

Idonotkno, 2022-01-21:
@EnderKi Unless your device is prototype.Otherwise, don't try Windows

EnderKi, 2022-01-21:
> 
It is prototype

EnderKi, 2022-01-21:
> @EnderKi Unless your device is prototype.Otherwise, don't try Windows

I want try it.

Idonotkno, 2022-01-21:
@EnderKi Follow the relevant links to enter the group.

EnderKi, 2022-01-21:
> @EnderKi Follow the relevant links to enter the group.

where can i find the group

Idonotkno, 2022-01-21:
@EnderKi I don't know.

BigfootACA, 2022-01-21:
https://forum.renegade-project.org/t/group-chat/68

closed 2022-01-23 by BigfootACA