edk2-porting/edk2-msm #93: driver for razerphone 2

whynot4444, 2022-01-31:
### Prerequisites

* [ ] Have you read the readme?
* [ ] Do you (want to) have your device in the supported list?
* [ ] Does the device have a Snapdragon 845 SOC?

### Description

Hello i have a razer phone 2, can i using driver and boot from one plus 6T ?, or i can try it 

**Expected behavior:** [What you expected to happen]

**Actual behavior:** [What actually happened]

BigfootACA, 2022-01-31:
https://forum.renegade-project.org/t/edk2-sdm845-new-device-porting-guide/417

whynot4444, 2022-02-01:
> https://forum.renegade-project.org/t/edk2-sdm845-new-device-porting-guide/417

thanks you, and WOA Driver can do like this ?

whynot4444, 2022-02-01:
> https://forum.renegade-project.org/t/edk2-sdm845-new-device-porting-guide/417

ok im try to complie, but it show unknow error, this razer phone 2 have 2 boot a and b, this tutorial can work ?

BigfootACA, 2022-02-01:
> > https://forum.renegade-project.org/t/edk2-sdm845-new-device-porting-guide/417
> 
> thanks you, and WOA Driver can do like this ?

based on sdm845-generic

whynot4444, 2022-02-02:
> > > https://forum.renegade-project.org/t/edk2-sdm845-new-device-porting-guide/417
> > 
> > 
> > ok im try to complie, but it show unknow error, this razer phone 2 have 2 boot a and b, this tutorial can work ?
> 
> yes

im try it show unknow error

whynot4444, 2022-02-03:
> > > https://forum.renegade-project.org/t/edk2-sdm845-new-device-porting-guide/417
> > 
> > 
> > ok im try to complie, but it show unknow error, this razer phone 2 have 2 boot a and b, this tutorial can work ?
> 
> yes

flash ok but hang logo, haizza

Idonotkno, 2022-02-03:
@whynot4444 Replace dtb, define resolution

whynot4444, 2022-02-03:
> @whynot4444 Replace dtb, define resolution

it only 1 dtb, resolution have 2 is 2K and fullhd, im choosing fullHD

Idonotkno, 2022-02-03:
@whynot4444 Replace common dtb and configuration, and compile common models.

whynot4444, 2022-02-03:
> @whynot4444 Replace common dtb and configuration, and compile common models.

ah you have tutorial of this, because this is first time im using linux

Idonotkno, 2022-02-03:
@whynot4444 Join the group

whynot4444, 2022-02-03:
> @whynot4444 Join the group

im on discord now

closed 2022-03-23 by BigfootACA