edk2-porting/edk2-msm #103: SD-Card does not work on installation (beryllium)

HeroRickyGAMES, 2022-02-21:
### Prerequisites

Não tenho um readme

### Description

I tried to install the system using a memory card but I couldn't find it in Windows Explorer or in Windows Disk Manager.
When installing the system I came across this.

Aro2142, 2022-02-22:
SD card does not seem to be supported yet and there is no information about it.

HeroRickyGAMES, 2022-02-23:
It would be very good if it worked because then it would give opportunities to make the dual boot work in a different way or more memory on the device since depending on the SD Card we would have more space to install programs, just like when we install some games on the pc but we have some secondary hard drives to run games or programs.

ClouDROID-VPN, 2022-03-14:
The cellular network/4g is working on poco f1 ?

HeroRickyGAMES, 2022-03-14:
It's working yes

ClouDROID-Dev, 2022-03-25:
> It would be very good if it worked because then it would give opportunities to make the dual boot work in a different way or more memory on the device since depending on the SD Card we would have [...]

Any update for SD Card on windows os its still not working? Because i have 64GB model and i want to install dual boot then only 18GB left both android and windows i want to insert my 32GB SD Card for more storage on Windows

HeroRickyGAMES, 2022-03-25:
> Any update for SD Card on windows os its still not working? Because i have 64GB model and i want to install dual boot then only 18GB left both android and windows i want to insert my 32GB SD Card [...]

no dude, it's not working. I tested it and it didn't work, I also wanted to install it on my 64gb SD card so the internal memory would stay intact but it's "almost impossible" unfortunately

ClouDROID-Dev, 2022-03-25:
But USB pendrive is working or portable hardrive ?

ClouDROID-Dev, 2022-03-28:
> > But USB pendrive is working or portable hardrive ?
> 
> USB is working properly! I didn't see any problems with USB. And the internal memory is like internal HD in Windows ARM, "any version"

Thanks because i have a harddrive maybe i can use it for more storage

HeroRickyGAMES, 2022-03-28:
> > > But USB pendrive is working or portable hardrive ?
> > 
> > 
> > USB is working properly! I didn't see any problems with USB. And the internal memory is like internal HD in Windows ARM, "any version"
> 
> Thanks because i have a harddrive maybe i can use it for more storage

Dude, you reminded me of something lol, if the touch was working, I could do that lol, before a while ago I wanted to connect a HD to my cell phone, but with Windows it would be better, Of course I managed to do it, but connect it with the case and use it normally, whether it's a notebook HD or SSD. Anyway, I still haven't managed to do that! But one day I will get it.

Still waiting for the Touch fix.

ClouDROID-Dev, 2022-03-29:
touch is working on fts panel but theres a bug for that

HeroRickyGAMES, 2022-03-29:
> touch is working on fts panel but theres a bug for that
> 
> you can see it there [edk2-porting/WOA-Drivers#25](https://github.com/edk2-porting/WOA-Drivers/issues/25)
> 
> i suggest to use a usb hub that have charging and for HDD/SSD or buy a wireless folding keyboard with touchpad model B033

I know, but mine is not a fts panel, so I can't use touch unfortunately ;-;

HeroRickyGAMES, 2022-03-29:
> touch is working on fts panel but theres a bug for that
> 
> you can see it there [edk2-porting/WOA-Drivers#25](https://github.com/edk2-porting/WOA-Drivers/issues/25)
> 
> i suggest to use a usb hub that have charging and for HDD/SSD or buy a wireless folding keyboard with touchpad model B033

If the touch is really working this will be my next investment, I analyzed it and yes there really was an advance in terms of Touch in Poco F1, now for it to be really usable it would be good a virtualization fix, in terms of battery drain this can be "fixed" using a lite version of windows, I made a windows ARM lite build (I'm brazilian so the video will be in Portuguese): https://youtube.com/shorts/3jLtyQn8YDY?feature=share

HeroRickyGAMES, 2022-03-29:
Another bug that I would be happy to see fixed is the memory card recognition... because I have a 64gb card that I use here.

ClouDROID-Dev, 2022-03-29:
How many storage left on your windows we are the same 64gb model are you using dual boot ?

HeroRickyGAMES, 2022-03-29:
> How many storage left on your windows we are the same 64gb model are you using dual boot ?

23gb left

HeroRickyGAMES, 2022-03-29:
> How many storage left on your windows we are the same 64gb model are you using dual boot ?

I've never tested dualboot, so I don't know if it would work because as far as I know it's only the 128gb model that is good with dualboot

ClouDROID-Dev, 2022-03-29:
> > How many storage left on your windows we are the same 64gb model are you using dual boot ?
> 
> 23gb left

Fresh install of windows how many left ?

HeroRickyGAMES, 2022-03-29:
> > > How many storage left on your windows we are the same 64gb model are you using dual boot ?
> > 
> > 
> > 23gb left
> 
> Fresh install of windows how many left ?

With the new one from scratch it's 24gb if I'm not mistaken.

ClouDROID-Dev, 2022-03-30:
> If the touch is really working this will be my next investment, I analyzed it and yes there really was an advance in terms of Touch in Poco F1, now for it to be really usable it would be good a [...]

are you using this lite version right now? this is english language ?

ClouDROID-Dev, 2022-04-05:
> If the touch is really working this will be my next investment, I analyzed it and yes there really was an advance in terms of Touch in Poco F1, now for it to be really usable it would be good a [...]

I installed this  but  i cant understand the language can i change it after setup

Am2648it, 2022-05-25:
Hey brother is it working properly ?

Am2648it, 2022-05-25:
And will you please tell me how can i install it on my POCO F1.
I won't be able to understand that how to install this .
Will you please tell me.

Am2648it, 2022-05-25:
@HeroRickyGAMES will you please help me .
Are you on any social media then please tell me.
We can discuss there .

Am2648it, 2022-05-25:
@HeroRickyGAMES hey brother are you there ?

Am2648it, 2022-05-25:
@ClouDROID-Dev  hey will you please tell me installation process i also have POCO F1. with unlocked bootloader 
Actually i unable to understand tha installation process .
Will you please tell me