edk2-porting/edk2-msm #106: xiaomi 8 issue Retrofit

wOscan, 2022-03-06:
hey,I have Retrofited my xiaomi 8 and i got a configure that is 10+256.
So,when i use your boot file,i got a blue screen and windows got some information to collect.

BigfootACA, 2022-03-23:
what is the blue screen code?

cander0815, 2022-03-29:
我安装了最新版本的镜像后，启动一直出现这个界面。
[image]

cander0815, 2022-03-29:
您好，我安装了最新版本的镜像后，启动一直出现这个界面。
[image]

Nigth01F0x, 2022-04-04:
Were you able to fix the problem?
once when I installed Windows 10 everything worked, but after reinstalling the blue screen

Nigth01F0x, 2022-04-04:
> 您好，我安装了最新版本的镜像后，启动一直出现这个界面。 [image]

Were you able to fix the problem?
once when I installed Windows 10 everything worked, but after reinstalling the blue screen

closed 2022-09-19 by strongtz