edk2-porting/edk2-msm #107: Oneplus 6 enchilada not charging while on Windows 11 (will provide any kind of logs details for debugging pupropses)

miIiano, 2022-03-06:
### Prerequisites

* [ ] Have you read the readme?
* [ ] Do you (want to) have your device in the supported list?
* [ ] Does the device have a Snapdragon 845 SOC?

### Description

[Description of the bug or feature]

**Expected behavior:** [What you expected to happen]

**Actual behavior:** [What actually happened]

miIiano, 2022-03-06:
It only works when i boot on OP firmware slot, and not charging at all while on Win 11.

fumeiusr, 2022-03-19:
IIRC OnePlus is very picky when it comes to chargers, it only allows certain chargers to work. Just try different chargers and see if you have any luck in getting it to charge in W11.

miIiano, 2022-03-20:
Tried over 10 chargers with diff cables also, still not allowing to charge and only way is booting back to Oxygen

Jbrimbelibap, 2022-03-20:
I am able to charge while in windows by plugging the phone to one of the usb ports on my pc. It charges quite slowly though

trojblue, 2022-04-25:
the 5v4a Nokia Lumia 950XL charger works for me with OP6. This one specifically https://www.ebay.ca/itm/184167639955

josc19, 2022-05-11:
I am also having this issue, it will pull current just fine in twrp and stock os and recovery mode. But when I load into windows its drawing 0 current. I tried hooking up my power bank (C to C) and it started charging the power bank???. I'm not sure if this is a new issue, but it definitely needs to be looked into. I confirmed all this with two different usb volt/amp/watt testers, with a to c and c to c. With a quick-change 3.0 brick and the charger that comes in the box, heck even my pc's dedicated charge port.

strongtz, 2022-05-11:
Could you try the images I sent in Telegram group just now? I'm not sure if it will work.
-------- 原始信息 --------发件人： josc19 ***@***.***> 日期: 2022/5/11  16:06  (GMT+08:00) 收件人： edk2-porting/edk2-sdm845 ***@***.***> 抄送： Subscribed ***@***.***> 主题： Re: [edk2-porting/edk2-sdm845] Oneplus 6 enchilada not charging while on Windows 11 (will provide any kind of logs details for debugging pupropses) (Issue #107) 
I am also having this issue, it will pull current just fine in twrp and stock os and recovery mode. But when I load into windows its drawing 0 current. I tried hooking up my power bank (C to C) and it started charging the power bank???. I'm not sure if this is a new issue, but it definitely needs to be looked into. I confirmed all this with two different usb volt/amp/watt testers, with a to c and c to c. With a quick-change 3.0 brick and the charger that comes in the box, heck even my pc's dedicated charge port.

strongtz, 2022-05-12:
Charging with a PD charger is supported now https://github.com/edk2-porting/edk2-sdm845/commit/f641c801b0d033e3a5a831671d1153d1f49cad1e

closed 2022-05-12 by strongtz