edk2-porting/edk2-msm #117: Windows Not charging

ClouDROID-Dev, 2022-04-06:
Windows OS is newly installed and have a default charger but still not charging even battery saver not showing

Aro2142, 2022-04-08:
The smartphone will be charged only from power supplies supporting QC 3.0 and higher!! or use the docking station.

ClouDROID-Dev, 2022-04-08:
> The smartphone will be charged only from power supplies supporting QC 3.0 and higher!! or use the docking station.

What you mean for this sir

Aro2142, 2022-04-09:
https://en.m.wikipedia.org/wiki/Quick_Charge

ClouDROID-Dev, 2022-04-13:
So it will charge if i change the adaptor with that ?

ClouDROID-Dev, 2022-04-13:
> https://en.m.wikipedia.org/wiki/Quick_Charge [image]

I used the orginal charger that you suggest but same problem is not charging on windows boot

Aro2142, 2022-04-14:
Your wire may not have a +/- data line.  
Check the connector, it may be damaged or dirty.
Charging works fine on my poco f1, but sometimes it may not work on the first try.

ClouDROID-Dev, 2022-04-14:
> 

Its charging on android boot but on windows is not i used dual boot on my poco F1

strongtz, 2022-05-11:
Use PD chargers instead of QC ones. The driver doesn't support QC currently.

closed 2022-05-11 by strongtz