edk2-porting/edk2-msm #118: POCO F1 USB HUB HDMI

ClouDROID-Dev, 2022-04-26:
I want to know if hdmi is working on windows arm on poco using this hub
[image]

Can someone here test the HDMI?

BigfootACA, 2022-04-27:
it's not possible to work. you need one that supports displaylink

ClouDROID-Dev, 2022-04-27:
> it's not possible to work. you need one that supports displaylink

I will wait for someone tried the hdmi on POCO F1

samzanemesis, 2022-04-27:
As stated by Bigfoot you need one that supports Displaylink, the display out alt modes are not configured on the poco hardware

alexander-pav, 2022-04-30:
Here is my Poco with DiplayLink hub
https://youtu.be/GHwdSv_D25Y

AdrianoA3, 2022-04-30:
@alexander-pav 1. What are the displaylink brand and model?
2. Does is charge the device?

alexander-pav, 2022-04-30:
> @alexander-pav 1. What are the displaylink brand and model? 2. Does is charge the device?
2 - No, it doesn't
We need additional OTG adapter with PD port but I don't know will it work with renegate windows or not

closed 2022-05-02 by BigfootACA