edk2-porting/edk2-msm #138: 坚果R1（trident）的v2.0rc2版本无法启动（BootImage is Incomplete）

ixuan789, 2022-07-20:
### Prerequisites

* [Y] Have you read the readme?
* [Y] Do you (want to) have your device in the supported list?
* [Y] Does the device have a Snapdragon 845 SOC?

### Description
坚果R1无法启动v2.0rc2的img镜像，尝试自行编译生成的镜像也无法启动。
2.0rc1版本可以正常启动。

**Expected behavior:** [What you expected to happen]

D:\> fastboot boot "boot-trident-v2.0rc1.img"
Sending 'boot.img' (1794 KB)                       OKAY [  0.052s]
Booting                                            OKAY [  0.039s]
Finished. Total time: 0.096s

**Actual behavior:** [What actually happened]

D:\> fastboot boot boot-trident-v2.0rc2.img
Sending 'boot.img' (4390 KB)                       OKAY [  0.118s]
Booting                                            FAILED (remote: 'BootImage is Incomplete')
fastboot: error: Command failed

strongtz, 2022-09-19:
rc2版本使用了`mkbootimg.py`来生成boot镜像以取代之前的`abootimg`，这样可以在受支持的设备上利用SimpleInit启动安卓。或许你可以尝试调整生成参数

troi12, 2022-11-11:
I have poco f3 and I dont know is it safe to install it on my phone?