edk2-porting/edk2-msm #139: Latest version of project does not compile

jodidas, 2022-08-09:
### Prerequisites

- [x] Have you read the readme?
- [x] Do you (want to) have your device in the supported list?
- [x] Does the device have a Snapdragon 845 SOC?

### Description

Environment: Ubuntu 20.04 WSL under Windows 11
The required dependencies have been installed in accordance with the compilation guide, but it still does not compile. The error is either an implicitly defined function or many proprietary structs and definitions such as EFI_STATUS cannot be found. Now I am baffled as to what the root cause of the problem is and have no way of getting around it.
My device is a Xiaomi dipper with SimpleInit enabled by default, but the errors occur mainly here.

[image]
[image]

KhanhNguyen9872, 2022-08-18:
what device you compile? i compile beryllium and polaris without any error

BigfootACA, 2022-08-27:
please cleanup your environment and recompile

closed 2022-09-12 by strongtz