edk2-porting/edk2-msm #143: Oneplus 6 Qualcomm crashdump on windows shutdown

gauravsuman007, 2022-08-25:
### Prerequisites

* [Yes ] Have you read the readme?
* [ Yes] Do you (want to) have your device in the supported list?
* [ Yes] Does the device have a Snapdragon 845 SOC?

### Description

Using the latest release edk2, everytime I shutdown windows 11 (latest updates already installed), my OnePlus 6 goes to Qualcomm crashdump mode. This happens every single time I shut down but doesn't occur when I do a restart.

I'm wondering if this is because windows 11 goes into some kind of hibernation when doing the shutdown and this is not supported? Using the restart would do a proper shut down - start cycle and so maybe it works ok. Just my theory.

**Expected behavior:** [Using shut down actually powers down the phone]

**Actual behavior:** [Phone goes into Qualcomm crashdump mode on shutdown]

txurtxil, 2022-08-26:
In windows 10 pro same problem.

gauravsuman007, 2022-08-26:
I can confirm that this is an issue with the windows 11 fast startup/hybrid sleep.

I disabled this feature in windows by using the following command in an elevated powershell
`powercfg -h off`

Now my phone shuts down without crashing. I'm still keeping this issue open until the dev team decides that they don't want to support fast startup/hybrid sleep

gauravsuman007, 2022-08-26:
Closed the issue by mistake, need feedback from the dev team before closing the issue

closed 2022-08-27 by BigfootACA