edk2-porting/edk2-msm #146: polaris(mix2s)使用最新版uefi蓝屏

Moe-hacker, 2022-09-10:
系统是mindows工具箱安装的win11，已经使用dism++的添加驱动功能添加了WOA-Drivers的release中最新的驱动，当我更新到最新版uefi(v2.0rc2)时，开机后锁屏界面由原来的横屏变为竖屏(重力传感器应该是工作了🧐)，随后还没输完密码就蓝屏报错SOC_SUBSYSTEM_FAILURE，一直是这样😥
我尝试过刷写最新的devcfg分区，但并不奏效。
但当我回退到v2.0rc1的boot，bug消失，不会蓝屏。
我觉得可能是重力传感器的锅，我待会儿试试删掉传感器驱动。

qaz6750, 2022-09-10:
这个蓝屏是子系统错误，如果升级驱动出现问题，请查看论坛的解决办法。

建议请使用英语，谢谢

Moe-hacker, 2022-09-10:
> 系统是mindows工具箱安装的win11，已经使用dism++的添加驱动功能添加了WOA-Drivers的release中最新的驱动，当我更新到最新版uefi(v2.0rc2)时，开机后锁屏界面由原来的横屏变为竖屏(重力传感器应该是工作了🧐)，随后还没输完密码就蓝屏报错SOC_SUBSYSTEM_FAILURE，一直是这样😥 我尝试过刷写最新的devcfg分区，但并不奏效。 但当我回退到v2.0rc1的boot，bug消失，不会蓝屏。 我觉得可能是重力传感器的锅，我待会儿试试删掉传感器驱动。

删除传感器驱动依然不起作用，或许我应该继续使用rc1😋

strongtz, 2022-09-19:
> 这个蓝屏是子系统错误，如果升级驱动出现问题，请查看论坛的解决办法。
> 
> 建议请使用英语，谢谢

github上用不用英语其实都可以的（

strongtz, 2022-09-19:
> > 系统是mindows工具箱安装的win11，已经使用dism++的添加驱动功能添加了WOA-Drivers的release中最新的驱动，当我更新到最新版uefi(v2.0rc2)时，开机后锁屏界面由原来的横屏变为竖屏(重力传感器应该是工作了monocle_face)，随后还没输完密码就蓝屏报错SOC_SUBSYSTEM_FAILURE，一直是这样disappointed_relieved 我尝试过刷写最新的devcfg分区，但并不奏效。 但当我回退到v2.0rc1的boot，bug消失，不会蓝屏。 我觉得可能是重力传感器的锅，我待会儿试试删掉传感器驱动。
> 
> 删除传感器驱动依然不起作用，或许我应该继续使用rc1yum

已经确认这个问题了，另外问一下，rc1没有问题吗？