edk2-porting/edk2-msm #204: BSOD while booting Windows 11

Docence, 2022-12-12:
### Prerequisites

* [✓] Have you read the wiki?
* [X] Is your device in the supported list?
* [✓] Does the device have a Snapdragon SOC which is supported in this repo?

### Description

**Expected behavior:** The device boots Windows

**Actual behavior:**Reboot twice in uefi then got blue screen, got the code "driver pnp watchdog"    I disabled driver signing and even I don't have any drivers installed.
My device is Razer Phone 2, uefi version 0.2.1

qaz6750, 2022-12-12:
- I suggest that you use Windbg to obtain some logs, so that you can determine the problem. At the same time, a correct ACPI is essential
- I also suggest you install some basic drivers, but 845 has not been added (started without using drivers),
- You should use the acpi of other devices. You need to modify one for your device, which includes many things that need to be modified. If you are only using it to start WinPE, you don't need to pay so much attention

qaz6750, 2022-12-12:
Every mobile phone device may have a unique problem, and it is necessary to continue to repair the ACPI/driver for the unique problem. If you are able to repair and debug it yourself, developers have their own things, lives, and devices they maintain. At the same time, it is difficult to repair an ACPI/driver without a device, and almost no one will do so

Docence, 2022-12-13:
> Every mobile phone device may have a unique problem, and it is necessary to continue to repair the ACPI/driver for the unique problem. If you are able to repair and debug it yourself, developers [...]

Thanks for your answer. I'm a beginner and I'm gradually learning these. I have another question, the uefi compiled with the latest repo (v2.0rc2) won't boot anyway, what could be wrong?

closed 2022-12-24 by qaz6750