edk2-porting/edk2-msm #210: 是否支持sdm710

BaichenXianchen, 2022-12-21:
### Prerequisites

* [ ] Have you read the wiki?
* [ ] Is your device in the supported list?
* [ ] Does the device have a Snapdragon SOC which is supported in this repo?

### Description

[Description of the bug or feature]

**Expected behavior:** [What you expected to happen]

**Actual behavior:** [What actually happened]

BaichenXianchen, 2022-12-21:
我的设备是Realme X（RMX1901）

qaz6750, 2022-12-21:
- If there is an xbl, extract the file from xbl, and then refer to the sdm660 platform for repair,

- If xbl wants you to send one

BaichenXianchen, 2022-12-21:
This is xbl of my device:
[xbl download](https://baichenxianchen-my.sharepoint.com/:u:/g/personal/baichenxianchen_baichenxianchen_onmicrosoft_com/EeZltDy7HvlDiUihqeLptFcB6hHe5_TV-7OSmNbheFEw1w?e=b7EyM9)

BaichenXianchen, 2022-12-21:
This is xbl.img of my device:
[xbl download](https://baichenxianchen-my.sharepoint.com/:u:/g/personal/baichenxianchen_baichenxianchen_onmicrosoft_com/EeZltDy7HvlDiUihqeLptFcB6hHe5_TV-7OSmNbheFEw1w?e=b7EyM9)

BaichenXianchen, 2022-12-21:
> * 如果有xbl，从xbl中解压文件，然后参考sdm660平台修复，
> * 如果 xbl 要你发送一个

这是我设备的xbl.img
[xbl.img下载](https://baichenxianchen-my.sharepoint.com/personal/baichenxianchen_baichenxianchen_onmicrosoft_com/_layouts/15/onedrive.aspx?id=%2Fpersonal%2Fbaichenxianchen%5Fbaichenxianchen%5Fonmicrosoft%5Fcom%2FDocuments%2Fxbl%2Eimg&parent=%2Fpersonal%2Fbaichenxianchen%5Fbaichenxianchen%5Fonmicrosoft%5Fcom%2FDocuments&ga=1)

BaichenXianchen, 2022-12-24:
我成功在我的手机上运行了EDK2，但是启动菜单无法通过音量键选择启动项

halal-beef, 2022-12-24:
You need to patch your buttonsdxe (extracted from xbl also)

qaz6750, 2022-12-24:
很高兴看到UEFI在你的手机上启动，对于你的问题：
- 1.SimpleInit 不需要patch也可以使用buttons，所以请添加buttons dxe到你的UEFI
- 2.如果需要使用SimpleInit 之外的功能，请patch buttons

另外欢迎贡献你所做的，如果想贡献可以尽量按以下要求谢谢
- 1.保证没有多余的文件
- 2.一个清晰的提交信息（这能为其它开发者提供便利）
- 3.尽量使用Git进行提交

closed 2022-12-24 by BaichenXianchen