edk2-porting/edk2-msm #216: Oneplus 6t recovery

Momoprawdziwe, 2022-12-31:
[image]
### Prerequisites

* [ ] Have you read the wiki?
* [ ] Is your device in the supported list?
* [ ] Does the device have a Snapdragon SOC which is supported in this repo?

### Description

[Description of the bug or feature]

[image]

TheMorc, 2022-12-31:
https://renegade-project.tech/en/install#h-44-your-pcdevice-needs-to-be-repaired

TheMorc, 2022-12-31:
What guide did you use and where did you get drivers from?

Momoprawdziwe, 2023-01-01:
https://youtu.be/CaQ9B6IGIjg 
This is guide I was using

TheMorc, 2023-01-01:
> https://youtu.be/CaQ9B6IGIjg
> This is guide I was using

Uh, we don’t provide any support for YouTube tutorials since they are bound to get outdated quickly as the project moves on.

closed 2023-01-01 by TheMorc