edk2-porting/edk2-msm #218: Fajita BSOD when compiling master branch

coreycusick, 2023-01-02:
### Prerequisites

* [y] Have you read the wiki?
* [y] Is your device in the supported list?
* [y] Does the device have a Snapdragon SOC which is supported in this repo?

### Description

I compiled the most recent build from master, for Fajita.

**Expected behavior:** [What you expected to happen]
Windows to boot normally

**Actual behavior:** [What actually happened]
Windows refuses to boot with the particular version reporting a mix of error codes, along with a few BSOD options, usually relating to SOC_SUBSYSTEM_FAILURE. I guess I’ve noticed devices getting ACPI updates, does something need to be done to the 6t config or the ACPI for the system to work as expected?

coreycusick, 2023-01-02:
I am aware the guide clearly states to use the v2.0rc2. I’m asking what differences are needed to allow the system to boot, as I’m going to attempt to fix it. The UEFI drivers also may be broken, as the touch screen doesn’t function, as well as it reporting an error if a button is pressed

TheMorc, 2023-01-03:
Oh.. I haven't touched WoA on 6/6T since 2021. One thing you can try is to go slowly between commits and check what has changed for fajita. Or maybe just try reverting ACPI to the same one from v2.0rc2.

coreycusick, 2023-01-03:
@TheMorc ok, i will try that. I’ve done a lot of work with ACPI tables for a different type of purpose. I’m actually kinda wondering why x64 gpu support doesn’t work for SD845, is that something that can be fixed or am I just gonna need to get a different device, I’m kinda interested in the Surface Duo anyways, for no other reason other than it looks cool. From my understanding it works on Snapdragon 855 series.

TheMorc, 2023-01-03:
> @TheMorc I’m actually kinda wondering why x64 gpu support doesn’t work for SD845, is that something that can be fixed or am I just gonna need to get a different device, I’m kinda interested in the Surface Duo anyways, for no other reason other than it looks cool. From my understanding it works on Snapdragon 855 series.

x64 GPU support relies on the GPU driver and this GPU driver doesn't support it. It's also not possible to port the 855 driver to 845.

qaz6750, 2023-01-06:
> ### 先决条件
> * [y] 你读过维基吗？
> * [y] 您的设备是否在受支持的列表中？
> * [y] 设备是否具有此存储库中支持的骁龙 SOC？
> 
> ### 描述
> 我为法吉塔编译了最新的主版本。
> 
> 预期**行为：[预期**发生的情况] Windows 正常启动
> 
> 实际**行为：[实际**发生了什么] Windows拒绝使用特定版本启动，报告混合错误代码以及一些BSOD选项，通常与SOC_SUBSYSTEM_FAILURE有关。我想我已经注意到设备正在获取 ACPI 更新，是否需要对 6t 配置或 ACPI 执行某些操作才能使系统按预期工作？

> 
- SOC_SUBSYSTEM_FAILURE：It is usually related to Qcsubsys, but in a few cases, it is related to other drivers