edk2-porting/edk2-msm #221: Oneplus 6t recovery

Momoprawdziwe, 2023-01-05:
[image]

I was using original renegade instructions

TheMorc, 2023-01-05:
where did you download drivers from and how is the driver zip called

Momoprawdziwe, 2023-01-05:
From renegade link 
Zip is named fajta

Momoprawdziwe, 2023-01-05:
Fajita*

Momoprawdziwe, 2023-01-12:
I installed few versions of drivers at once to check if it gonna work 
[image]

Momoprawdziwe, 2023-01-12:
[image]

Momoprawdziwe, 2023-01-12:
[image]

Momoprawdziwe, 2023-01-23:
I installed sdm845 generic drivers first then I added 6t drivers and it worked