edk2-porting/edk2-msm #222: [Request] Port for Mi 11 Lite 4G 732G (SM7150)

ask9027, 2023-01-08:
### Prerequisites

* [X] Have you read the wiki?
* [ ] Is your device in the supported list?
* [X] Does the device have a Snapdragon SOC which is supported in this repo?

### Description

Since Redmi note 10 pro and Mi 11 Lite 4G have same CPU Config.
Both have SM7150( Snapdragon 732G)
Please make port for Mi 11 Lite 4G (Board : courbetin or courbet)

Here is the Details of CPU.
[image]

qaz6750, 2023-01-08:
You should know： 
- Developers don't have so much time, and debugging without devices is extremely difficult
- Since it is the same soc, you can try to transplant it yourself

ask9027, 2023-01-08:
Okay, i will try myself. Thanks

closed 2023-01-08 by ask9027