edk2-porting/edk2-msm #230: could you support more devices ?

superiums, 2023-01-18:
### Prerequisites

* [ Y ] Have you read the wiki?
* [ N ] Is your device in the supported list?
* [ N ] Does the device have a Snapdragon SOC which is supported in this repo?

### Description

[support more devices]

really Great Project!

i'm on motorola edge s (nio).
This is the [kernel code](https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RRN31.Q1-20-52-3)
This is the [device information](https://wiki.lineageos.org/devices/nio/)

qaz6750, 2023-01-19:
- I suggest you try your own transplantation

MollySophia, 2023-01-24:
Actually the more devices this project supports, the harder it is to be maintained. The reason is that there are not enough developers, and developers cannot own that many devices to port and test with convenience.
BTW Snapdragon 865/870 really doesn't have much support now :P

superiums, 2023-01-27:
> * I suggest you try your own transplantation

is there any guid document ?

qaz6750, 2023-01-28:
> > * I suggest you try your own transplantation
> 
> is there any guid document ?

* https://renegade-project.tech/en/porting/device

closed 2023-02-19 by qaz6750