edk2-porting/edk2-msm #248: Build for New adding device winnerlte

Ost268, 2023-01-28:
(Python 3.8.10 on linux) Traceback (most recent call last):
  File "/home/ost268/edk2-msm/Common/edk2/BaseTools/BinWrappers/PosixLike/../../Source/Python/build/build.py", line 2695, in Main
    MyBuild.Launch()
  File "/home/ost268/edk2-msm/Common/edk2/BaseTools/BinWrappers/PosixLike/../../Source/Python/build/build.py", line 2490, in Launch
    self._MultiThreadBuildPlatform()
  File "/home/ost268/edk2-msm/Common/edk2/BaseTools/BinWrappers/PosixLike/../../Source/Python/build/build.py", line 2282, in _MultiThreadBuildPlatform
    Wa, self.BuildModules = self.PerformAutoGen(BuildTarget,ToolChain)
  File "/home/ost268/edk2-msm/Common/edk2/BaseTools/BinWrappers/PosixLike/../../Source/Python/build/build.py", line 2133, in PerformAutoGen
    Wa = WorkspaceAutoGen(
  File "/home/ost268/edk2-msm/Common/edk2/BaseTools/Source/Python/AutoGen/WorkspaceAutoGen.py", line 43, in __init__
    self._InitWorker(Workspace, MetaFile, Target, Toolchain, Arch, *args, **kwargs)
  File "/home/ost268/edk2-msm/Common/edk2/BaseTools/Source/Python/AutoGen/WorkspaceAutoGen.py", line 116, in _InitWorker
    self.ProcessModuleFromPdf()
  File "/home/ost268/edk2-msm/Common/edk2/BaseTools/Source/Python/AutoGen/WorkspaceAutoGen.py", line 212, in ProcessModuleFromPdf
    if self.FdfProfile:
  File "/home/ost268/edk2-msm/Common/edk2/BaseTools/Source/Python/Common/caching.py", line 28, in __get__
    Value = obj.__dict__[self._function.__name__] = self._function(obj)
  File "/home/ost268/edk2-msm/Common/edk2/BaseTools/Source/Python/AutoGen/WorkspaceAutoGen.py", line 188, in FdfProfile
    Fdf.ParseFile()
  File "/home/ost268/edk2-msm/Common/edk2/BaseTools/Source/Python/GenFds/FdfParser.py", line 1320, in ParseFile
    while self._GetFd() or self._GetFv() or self._GetFmp() or self._GetCapsule() or self._GetRule() or self._GetOptionRom():
  File "/home/ost268/edk2-msm/Common/edk2/BaseTools/Source/Python/GenFds/FdfParser.py", line 2116, in _GetFv
    isFile = self._GetFileStatement(FvObj)
  File "/home/ost268/edk2-msm/Common/edk2/BaseTools/Source/Python/GenFds/FdfParser.py", line 2564, in _GetFileStatement
    self._GetFilePart(FfsFileObj)
  File "/home/ost268/edk2-msm/Common/edk2/BaseTools/Source/Python/GenFds/FdfParser.py", line 2660, in _GetFilePart
    raise Warning.ExpectedCurlyClose(self.FileName, self.CurrentLineNumber)
GenFds.FdfParser.Warning: expected '}' near line 6, column 18: "ButtonsDxe"

- Failed -
Build end time: 22:17:29, Jan.27 2023
Build total time: 00:00:01

+ return 1
root@ost268-VirtualBox:/home/ost268/edk2-msm#

BigfootACA, 2023-01-28:
looks like a syntax error?

Ost268, 2023-01-28:
Active Platform          = /home/ost268/edk2-msm/Platform/Samsung/sm8150/winnerlte.dsc
 

build.py...
 : error C0DE: Unknown fatal error when processing [/home/ost268/edk2-msm/Platform/Samsung/sm8150/winnerlte.dsc]
	
(Please send email to devel@edk2.groups.io for help, attaching following call stack trace!)

(Python 3.8.10 on linux) Traceback (most recent call last):
  File "/home/ost268/edk2-msm/Common/edk2/BaseTools/BinWrappers/PosixLike/../../Source/Python/build/build.py", line 2695, in Main
    MyBuild.Launch()
  File "/home/ost268/edk2-msm/Common/edk2/BaseTools/BinWrappers/PosixLike/../../Source/Python/build/build.py", line 2490, in Launch
    self._MultiThreadBuildPlatform()
  File "/home/ost268/edk2-msm/Common/edk2/BaseTools/BinWrappers/PosixLike/../../Source/Python/build/build.py", line 2282, in _MultiThreadBuildPlatform
    Wa, self.BuildModules = self.PerformAutoGen(BuildTarget,ToolChain)
  File "/home/ost268/edk2-msm/Common/edk2/BaseTools/BinWrappers/PosixLike/../../Source/Python/build/build.py", line 2133, in PerformAutoGen
    Wa = WorkspaceAutoGen(
  File "/home/ost268/edk2-msm/Common/edk2/BaseTools/Source/Python/AutoGen/WorkspaceAutoGen.py", line 43, in __init__
    self._InitWorker(Workspace, MetaFile, Target, Toolchain, Arch, *args, **kwargs)
  File "/home/ost268/edk2-msm/Common/edk2/BaseTools/Source/Python/AutoGen/WorkspaceAutoGen.py", line 116, in _InitWorker
    self.ProcessModuleFromPdf()
  File "/home/ost268/edk2-msm/Common/edk2/BaseTools/Source/Python/AutoGen/WorkspaceAutoGen.py", line 212, in ProcessModuleFromPdf
    if self.FdfProfile:
  File "/home/ost268/edk2-msm/Common/edk2/BaseTools/Source/Python/Common/caching.py", line 28, in __get__
    Value = obj.__dict__[self._function.__name__] = self._function(obj)
  File "/home/ost268/edk2-msm/Common/edk2/BaseTools/Source/Python/AutoGen/WorkspaceAutoGen.py", line 188, in FdfProfile
    Fdf.ParseFile()
  File "/home/ost268/edk2-msm/Common/edk2/BaseTools/Source/Python/GenFds/FdfParser.py", line 1320, in ParseFile
    while self._GetFd() or self._GetFv() or self._GetFmp() or self._GetCapsule() or self._GetRule() or self._GetOptionRom():
  File "/home/ost268/edk2-msm/Common/edk2/BaseTools/Source/Python/GenFds/FdfParser.py", line 2116, in _GetFv
    isFile = self._GetFileStatement(FvObj)
  File "/home/ost268/edk2-msm/Common/edk2/BaseTools/Source/Python/GenFds/FdfParser.py", line 2564, in _GetFileStatement
    self._GetFilePart(FfsFileObj)
  File "/home/ost268/edk2-msm/Common/edk2/BaseTools/Source/Python/GenFds/FdfParser.py", line 2660, in _GetFilePart
    raise Warning.ExpectedCurlyClose(self.FileName, self.CurrentLineNumber)
GenFds.FdfParser.Warning: expected '}' near line 6, column 18: "ButtonsDxe"

- Failed -
Build end time: 16:50:49, Jan.28 2023
Build total time: 00:00:01

Ost268, 2023-01-28:
https://github.com/Ost268/edk2-msm
https://github.com/Ost268/edk2-msm-binary

qaz6750, 2023-01-28:
I think the problem may be here https://github.com/Ost268/edk2-msm/blob/master/Platform/Samsung/sm8150/winnerlte.fdf.inc

Ost268, 2023-01-28:
Building ... /home/ost268/edk2-msm/Common/edk2/MdeModulePkg/Application/UiApp/UiApp.inf [AARCH64]
/usr/bin/aarch64-linux-gnu-ld: помилка: Optimization level must be between 0 and 3
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [GNUmakefile:415: /home/ost268/edk2-msm/workspace/Build/winnerlte/RELEASE_CLANG38/AARCH64/MdeModulePkg/Application/UiApp/UiApp/DEBUG/UiApp.dll] error 1

Ost268, 2023-01-28:
Building ... /home/ost268/edk2-msm/Common/edk2/MdeModulePkg/Application/UiApp/UiApp.inf [AARCH64]
/usr/bin/aarch64-linux-gnu-ld: error: Optimization level must be between 0 and 3
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [GNUmakefile:415: /home/ost268/edk2-msm/workspace/Build/winnerlte/RELEASE_CLANG38/AARCH64/MdeModulePkg/Application/UiApp/UiApp/DEBUG/UiApp.dll] error 1

qaz6750, 2023-01-28:
Please refer to https://github.com/edk2-porting/edk2-msm/pull/197/commits/ce907c749a77abdb7ccc4d18d50f2a9f110b10e8

Ost268, 2023-01-28:
Active Platform          = /home/ost268/edk2-msm/Platform/Samsung/sm8150/winnerlte.dsc

build.py...
/home/ost268/edk2-msm/Platform/Samsung/sm8150/winnerlte.dsc(...): error 4000: Instance of library class [SerialPortLib] is not found
	in [/home/ost268/edk2-msm/Common/edk2/MdePkg/Library/BaseDebugLibSerialPort/BaseDebugLibSerialPort.inf] [AARCH64]
	consumed by module [/home/ost268/edk2-msm/GPLDrivers/Library/SimpleInit/src/main/SimpleInitMain.inf]
 

- Failed -
Build end time: 17:16:31, Jan.28 2023
Build total time: 00:00:01

+ return 1

Ost268, 2023-01-28:
Building ... /home/ost268/edk2-msm/Common/edk2/MdeModulePkg/Application/UiApp/UiApp.inf [AARCH64]
/usr/bin/aarch64-linux-gnu-ld: error: Optimization level must be between 0 and 3
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [GNUmakefile:415: /home/ost268/edk2-msm/workspace/Build/winnerlte/RELEASE_CLANG38/AARCH64/MdeModulePkg/Application/UiApp/UiApp/DEBUG/UiApp.dll] error1

build.py...
 : error 7000: Failed to execute command
	make -s tbuild [/home/ost268/edk2-msm/workspace/Build/winnerlte/RELEASE_CLANG38/AARCH64/MdeModulePkg/Application/UiApp/UiApp]

build.py...
 : error 7000: Failed to execute command
	make -s tbuild [/home/ost268/edk2-msm/workspace/Build/winnerlte/RELEASE_CLANG38/AARCH64/MdeModulePkg/Bus/Usb/UsbMassStorageDxe/UsbMassStorageDxe]

build.py...
 : error F002: Failed to build module
	/home/ost268/edk2-msm/Common/edk2/MdeModulePkg/Application/UiApp/UiApp.inf [AARCH64, CLANG38, RELEASE]

- Failed -
Build end time: 17:48:19, Jan.28 2023
Build total time: 00:00:29

+ return 1

Ost268, 2023-01-28:
Not work

Ost268, 2023-01-28:
> (Python 3.8.10 on linux) Traceback (most recent call last): File "/home/ost268/edk2-msm/Common/edk2/BaseTools/BinWrappers/PosixLike/../../Source/Python/build/build.py", line 2695, in Main [...]

Ost268, 2023-01-28:
Build error for any device

qaz6750, 2023-01-28:
[image]
* Obviously you need to use --fixclang
* At the same time, the construction error of the new device should be caused by your own misoperation. I see that you have introduced many useless files. Please also pay attention to the syntax problem,
[image]
* At the same time, you don't understand what I mean, just copy it blindly, and don't see that it can be used directly
* https://github.com/Ost268/edk2-msm/commit/57edfe01c1c50b008ef280aa0aac3918f3bdcd49  You even changed most of the memory mappings, which is likely to cause problems in other devices. You should not change so much
* There is no problem with the UEFI startup of r5q. You can refer to r5q and migrate according to the migration guide

Ost268, 2023-01-28:
Fd File Name:SM8150_UEFI (/home/ost268/edk2-msm/workspace/Build/winnerlte/RELEASE_CLANG38/FV/SM8150_UEFI.fd)

Generate Region at Offset 0x0
   Region Size = 0x1000000
   Region Name = FV

Generating FVMAIN_COMPACT FV

Generating FVMAIN FV
########################################
########################################
########################################
##############
GUID cross reference file can be found at /home/ost268/edk2-msm/workspace/Build/winnerlte/RELEASE_CLANG38/FV/Guid.xref

FV Space Information
FVMAIN [99%Full] 33987456 (0x2069b80) total, 33987448 (0x2069b78) used, 8 (0x8) free
FVMAIN_COMPACT [40%Full] 16777216 (0x1000000) total, 6751816 (0x670648) used, 10025400 (0x98f9b8) free

- Done -
Build end time: 20:22:12, Jan.28 2023
Build total time: 00:02:03

+ _call_hook platform_build_kernel
+ local NAME=platform_build_kernel
+ shift
+ declare -F platform_build_kernel
+ eval platform_build_kernel
++ platform_build_kernel
++ cat /home/ost268/edk2-msm/tools/BootShim/BootShim.bin /home/ost268/edk2-msm/workspace/Build/winnerlte/RELEASE_CLANG38/FV/SM8150_UEFI.fd
++ '[' 1 = 0 ']'
++ gzip -c
++ cat /home/ost268/edk2-msm/workspace/uefi-winnerlte.img.gz /home/ost268/edk2-msm/Platform/Samsung/sm8150/FdtBlob_compat/winnerlte.dtb
++ cp /home/ost268/edk2-msm/workspace/uefi-winnerlte.img.gz-dtb /home/ost268/edk2-msm/workspace/uefi-winnerlte-kernel
+ return 0
+ _call_hook platform_build_bootimg
+ local NAME=platform_build_bootimg
+ shift
+ declare -F platform_build_bootimg
+ eval platform_build_bootimg
++ platform_build_bootimg
++ python3 /home/ost268/edk2-msm/tools/mkbootimg.py --kernel /home/ost268/edk2-msm/workspace/uefi-winnerlte-kernel --ramdisk ramdisk --kernel_offset 0x00000000 --ramdisk_offset 0x00000000 --tags_offset 0x00000000 --os_version 12.0.0 --os_patch_level 2022-11 --header_version 1 -o /home/ost268/edk2-msm/boot-winnerlte.img
+ return 0
+ echo 'Build done: /home/ost268/edk2-msm/boot-winnerlte.img'
Build done: /home/ost268/edk2-msm/boot-winnerlte.img
+ set +x

Ost268, 2023-01-28:
boot after build, memory map failed
[Image](https://drive.google.com/file/d/1iX1ul4nw-mZqotdL2w6E50RHOVkD508S/view?usp=share_link)

closed 2023-01-28 by Ost268