edk2-porting/edk2-msm #259: Not working boot uefi for winnerlte

Ost268, 2023-02-04:
after compiling the kernel, does not load in uefi
[image]

Ost268, 2023-02-05:
[image]
Crash uefi memory map

Ost268, 2023-02-05:
[image]
Build working in MU-SM8150PKG.Maybe something's wrong with acpi?but i don't know how to fix it

TheMorc, 2023-02-05:
> [image]
> Build working in MU-SM8150PKG.Maybe something's wrong with acpi?but i don't know how to fix it

You just missed this:
https://renegade-project.tech/en/Troubleshooting#h-4-your-pcdevice-needs-to-be-repaired

Ost268, 2023-02-05:
Usb c port not working

TheMorc, 2023-02-05:
> Usb c port not working

How did you get Windows, EFI and drivers there then lol?

Ost268, 2023-02-05:
Not input usb Mouse and keyboard

TheMorc, 2023-02-05:
What does that have to do with driver signatures being enabled

Ost268, 2023-02-05:
I du, but i can't connect anything

Ost268, 2023-02-05:
how should I type commands in cmd, by telekinesis

leandrofriedrich, 2023-02-06:
tried with a powered hub yet?

Ost268, 2023-02-06:
[image]
such

qaz6750, 2023-02-09:
* I think now you can start uefi normally. This problem should be closed

Ost268, 2023-02-09:
which

Ost268, 2023-02-09:
I think that acpi table

Ost268, 2023-02-09:
it is difficult to know about the problem if the log is empty