edk2-porting/edk2-msm #263: Cannot boot Windows 11 22H2 in Polaris

dandingdjianjiao, 2023-02-09:
### Prerequisites

* [yes] Have you read the wiki?
* [yes] Is your device in the supported list?
* [yes] Does the device have a Snapdragon SOC which is supported in this repo?

### Description
I use the newest released boot image (version 2.0rc2) to boot Windows 11 22H2 on my polaris device ,and it crushed with error "System Thread Exception not Handled" . I flashed the devcfg partition but it dont work.If I use the older boot image(version 2.0rc1),it can boot properly.I dont know the reason.

qaz6750, 2023-02-09:
* Try to use 2.0rc2. At present, this problem has appeared on Mi8 and Mix2s. It may take some time to investigate the cause

dandingdjianjiao, 2023-02-09:
> 

Actually, I used 2.0rc2 first,it crushed.

qaz6750, 2023-02-26:
* This question is repeated with #257
* This is a known problem, but no one fixes & maintains it