edk2-porting/edk2-msm #269: Dualboot stuck at SimpleInit on Xiaomi Pad 5

wochgn, 2023-02-19:
### Prerequisites

* [√] Have you read the wiki?
* [√] Is your device in the supported list?
* [√] Does the device have a Snapdragon SOC which is supported in this repo?

### Description

[Description of the bug or feature]

**Expected behavior:** [What you expected to happen]When selected "Reboot to Android", the system reboots to Android

**Actual behavior:** [What actually happened]When selected "Reboot to Android", the system rebooted but entered Simpleinit again.

The device that couldn't dualboot is Xiaomi Pad 5 with MIUI 14 for Pad V14.0.23.2.13.DEV. However, on another Xiaomi Pad 5 with MIUI 14 for Pad V14.0.23.1.9.DEV, it works perfectly.

leandrofriedrich, 2023-02-19:
@wochgn lol

qaz6750, 2023-02-19:
* In fact, Xiaomi 9 also cannot work on some specific versions, and I think it should not be the reason for Android kernel, which is more like a system problem. At the same time, all this is just speculation, because we can't know what every ROM developer has done

wochgn, 2023-02-20:
On my Xiaomi Pad 5 running MIUI 14 for Pad V14.23.1.9.DEV, the dualboot stopped working and I was also stuck at Simpleinit...(While it did work yesterday)

J0SH1X, 2023-02-23:
@YiyiPenguin Its known since HTC HD2 days that booting windows will remove power off charging lol

YiyiPenguin, 2023-02-23:
After several tests, it was broken under Arrow OS 13

wochgn, 2023-02-23:
Here are the logs from UEFI.

[UefiLog1.txt](https://github.com/edk2-porting/edk2-msm/files/10814070/UefiLog1.txt)
[UefiLog0.txt](https://github.com/edk2-porting/edk2-msm/files/10814071/UefiLog0.txt)
[UefiLog2.txt](https://github.com/edk2-porting/edk2-msm/files/10814072/UefiLog2.txt)
[UefiLog3.txt](https://github.com/edk2-porting/edk2-msm/files/10814073/UefiLog3.txt)
[UefiLog4.txt](https://github.com/edk2-porting/edk2-msm/files/10814074/UefiLog4.txt)

MollySophia, 2023-02-26:
> Hey, I have already tested this method under Arrow OS 13, it works fine. To test the stability of this new method, I restarted my device for about 7 times, without stucking in the fastboot or any behaviour out of control. I think the main reason of the method failed under miui because miui has the self protect, it will recovery the kernel settings, which caused the device can only boot into windows rather than miui. (Emmmmm, finally, if the uefi support poweroff charging, it will be better.) Sorry for my bad English......

Actually on my mipad 5 it works well with miui 14. This method seems to have some natural drawbacks that prevent it to be working on every device, and I haven't found a solution yet...

qaz6750, 2023-02-26:
> > Hey, I have already tested this method under Arrow OS 13, it works fine. To test the stability of this new method, I restarted my device for about 7 times, without stucking in the fastboot or any behaviour out of control. I think the main reason of the method failed under miui because miui has the self protect, it will recovery the kernel settings, which caused the device can only boot into windows rather than miui. (Emmmmm, finally, if the uefi support poweroff charging, it will be better.) Sorry for my bad English......
> 
> Actually on my mipad 5 it works well with miui 14. This method seems to have some natural drawbacks that prevent it to be working on every device, and I haven't found a solution yet...
> 
> Sorry for my late response

It seems to be true that even though I worked well in my mi9 before, I and it can't work now. Maybe it needs some investigation and improvement

hiprivsid, 2024-06-04:
> Any advice on a custom ROM or something to make this "Reboot to Android" option work? With official HyperOS Global it does not work on my Xiaomi Pad 5 😭

project is dead🙏🙏🙏

graphiks, 2024-06-04:
@hiprivsid hyper-vsid

palexdev, 2024-06-04:
@MollySophia any news on this?
Seems like my device has the same issue. Linux + EvoX