edk2-porting/edk2-msm #293: Device fails to boot with synchronous exception with OP 7T Pro (hotdog)

jurasek05, 2023-05-17:
### Prerequisites

* [ Yes] Have you read the wiki?
* [ Yes] Is your device in the supported list?
* [Yes ] Does the device have a Snapdragon SOC which is supported in this repo?

### Description
Oneplus 7T Pro (hotdog)
Using 2301.1 release of boot-hotdog.img

**Expected behavior:**
Normal function

**Actual behavior:**
After windows for arm install I successfully booted to windows, but after several minutes phone restarted and end up with these error message: https://cdn.discordapp.com/attachments/736563593616556134/1108417567590187100/20230517_173524.jpg
This message shows just after I booted with image. Shows after windows install and first run, before it everything was fine. Restarted phone many times, but didnt help. This was my 4. attempt to make it run, but still same result.

maalos, 2023-05-20:
For me this issue happens when trying to boot the UEFI image with fastboot on lisa (exact same error message)

golamdastagir, 2023-06-24:
Device: Oneplus 7T(hotdog)
Also using the 2301.1 release of "boot-hotdog.img"

Did you face any screen flicker issues after booting "boot-hotdog.img"? 
I cannot carry on with the installation process due to the screen flickering.

realcosmo123, 2023-07-17:
just delete simpleinit.cfg on other partitions with linux running