edk2-porting/edk2-msm #294: OnePlus 5t (dumpling) Volume/Power keys don't work in UEFI

wolfinabox, 2023-05-20:
### Prerequisites

* [x] Have you read the wiki?
* [x] Is your device in the supported list?
* [x] Does the device have a Snapdragon SOC which is supported in this repo?

### Description
Volume/Power keys on OnePlus 5t (dumpling) don't work in the UEFI boot menu, which prevents navigating the menu.
(No messages/errors show upon pressing, there's just no response. The device automatically boots into the EFI shell after the timeout)
Tried release [2301.1](https://github.com/edk2-porting/edk2-msm/releases/tag/2301.1) as well as compiling 880e0104bc35a182b88c7ad81c2b0f843aa5d202 from source, no dice.
The volume/power keys do still work in Android, and the power button works to go past the "This boot loader is unlocked" screen before booting, just not in the edk2-msm UEFI.
Wiki says [UEFI buttons should be supported](https://renegade-project.tech/en/state#supported-device-features-on-windows), so not sure if I've done something wrong, or there's an issue.

Happy to test/provider any other info that's needed, as I'd love to get this running on my old phone.

qaz6750, 2023-06-11:
* We need to fix buttondxe for it, which can be referenced using Op5

shenghuang147, 2023-12-07:
> * We need to fix buttondxe for it

I went straight to Cheeseburger's ButtonsDxe and the buttons worked fine.

Ok12340, 2024-06-18:
> > * We need to fix buttondxe for it
> 
> I used Cheeseburger's ButtonsDxe and the buttons worked fine.

Could you tell me how to do it please

shenghuang147, 2024-06-18:
> Could you tell me how to do it please

I'm sorry, because the time is too far in the past. I'm not sure if I'll forget some details.

As far as I remember, all you need to do is copy the content from 
https://github.com/edk2-porting/edk2-msm/blob/81e670bf6a92a8f7044b393012423f2e72ad8c43/Platform/Oneplus/msm8998/cheeseburger.fdf.inc#L2-L6

to

https://github.com/edk2-porting/edk2-msm/blob/81e670bf6a92a8f7044b393012423f2e72ad8c43/Platform/Oneplus/msm8998/dumpling.fdf.inc

Ok12340, 2024-06-18:
> I'm sorry, because the time is too far in the past. I'm not sure if I'll forget some details.
>
> As far as I remember, all you need to do is copy the content from [...]

And what should i do after that. I have to get the .img file somehow

Daniel224455, 2024-06-18:
have you tried edk2-msm8998