edk2-porting/edk2-msm #300: sm8250 edk2 support

dabao1955, 2023-06-04:
### Prerequisites

* [x] Have you read the wiki?
* [ ] Is your device in the supported list?
* [ ] Does the device have a Snapdragon SOC which is supported in this repo?

### Description

support sm8250 device for edk2，just run linux.

AdrianoA3, 2023-06-04:
@dabao1955 Does your device have mainlined kernel?

dabao1955, 2023-06-04:
> @dabao1955 Does your device have mainlined kernel?

What do you mean?

qaz6750, 2023-06-11:
* In fact, the Sm8250 SOC has already been supported, and now only EDk2 needs to be ported and tested for the device. Additionally, I don't think it makes much sense

ungeskriptet, 2023-06-18:
> > * In fact, the Sm8250 SOC has already been supported, and now only EDk2 needs to be ported and tested for the device. Additionally, I don't think it makes much sense
> 
> I don't think so.Although sm8250 devices can not run windows.

It is supported. Look in the Silicon and Platform folder. You can boot Android and Linux with it.

SM8250 device can boot very minimal WinPE but it's not really useful. If you want to add support for an unsupported device you can follow [this commit](https://github.com/edk2-porting/edk2-msm/commit/fcd382a2815e0e5286a0c4bc150c9f070e270da3)

ungeskriptet, 2023-06-24:
> > > > * In fact, the Sm8250 SOC has already been supported, and now only EDk2 needs to be ported and tested for the device. Additionally, I don't think it makes much sense
> > > 
> > > 
> > > I don't think so.Although sm8250 devices can not run windows.
> > 
> > 
> > It is supported. Look in the Silicon and Platform folder. You can boot Android and Linux with it.
> > SM8250 device can boot very minimal WinPE but it's not really useful. If you want to add support for an unsupported device you can follow [this commit](https://github.com/edk2-porting/edk2-msm/commit/fcd382a2815e0e5286a0c4bc150c9f070e270da3)
> 
> This submission only has one Lenovo device, my device is Xiaomi Mi 10pro, I was hoping to try to boot Linux via UEFI on this device, but I didn't find a channel to download and a way to build.

As I said, you can follow this commit as an example to add support for your Mi 10 Pro. You can pretty much copy the j716f port and replace the values that are device specific (like memory map)