edk2-porting/edk2-msm #309: Support for SDM429

Minionguyjpro, 2023-07-20:
Please add support for this SoC. After being added, I can add support for my SM-T290 tablet (Samsung Galaxy Tab A 8.0 2019). It is not as powerful as other ones, but it might be able to run some Linux distributions like Xubuntu.

halal-beef, 2023-07-20:
@Minionguyjpro lmfao

nedokaka, 2023-08-14:
@koke228 КУКИ ТЫ ЕБАНАТ БЛЯТЬ

rxuglr, 2023-08-14:
@nedokaka ты нахуя на русском написал

ASDownloadingAccount, 2024-07-31:
You can [port this yourself.](https://renegade-project.tech/en/porting)

zxcwsurx, 2024-07-31:
Hui penis pizda

Minionguyjpro, 2024-08-01:
> You can [port this yourself.](https://renegade-project.tech/en/porting)

This is not completely true. Sure it says porting, but this repository needs to have a device configuration for it, which it doesn't have. So I think someone needs to add a configuration and EFI drivers in this repository before I can really go over and use and build it here.