edk2-porting/edk2-msm #315: poco f1 (beryllium) nvt-ts touch screen stop working after lock screen

sdyby2006, 2023-08-03:
### Prerequisites

* [true] Have you read the wiki?
* [true] Is your device in the supported list?
* [true] Does the device have a Snapdragon SOC which is supported in this repo?

### Description

[Description of the bug or feature]

**Expected behavior:** [What you expected to happen]

**Actual behavior:** [What actually happened]

HergiCz, 2024-02-09:
Hello I have just small hints to this topic as I am trying hard to somehow fix it.

1) I have tried for all drivers to disable following
"Allow the computer to turn of this device to save power"
+ turning of thebfast startup in power options

But no luck

2) I have tried to disable GPU driver (Adreno 680) which on its own leads to issue. Than touch screen stops working. Re enable of driver doesnt help (ctrl shift windows key + B) doesnt helps. Only what helps is restart

3) lock screen leads to same problems as point 2.

Small and only help for me was to disable GPU and restart. Then touch actually works even if

a) lock and unlock screen
b) turns on GPU driver

I think the issue might be somehow  in GPU driver... Isnt a problem that pocophone actually has adreno 630 gpu? I am not sure how to proceed further but maybe this will help to recognize something more. I would really glad if this would get solved. For now I will be using it withou GPU as my applications mostly doesnt require GPU processing. 

Note SDI tool actually recognizes this too and I could also download a driver for 630 gpu. But display inmidiatly gets mega dimm and GPU is not working (i thin 43 error code at driver) also leads to some BSOD sometimes upon boot... 

But SDI tool actually recognized some drivers i thing for bus, and cameras... Not sure if anybody knows and not sure if it help to something, just saying... 

I am using it for car tuning which needs specific windows sw which is not GPU extensive.

HergiCz, 2024-02-09:
I have just find very interesting fact.

If driver Focaltech 5x digitizer (ft8607) is disabled. Touch will stop work if enabled touch will start work as everybody would thougt. But if has been disabled befure user disable gpu driver or lock screen or turn of the screen, it can be than reenabled. AND WHOA touch will be working!

So we have already two small not very perfect workarounds :)

Maybe this can be somehow used already to write some script or what so ever... However fix in touchscreen driver would be better way ofc...

Thank you for any ideas!

HergiCz, 2024-02-10:
OK i could manage work around this problems by scrips.
In short and simply put

1) Its possible to create a script which will turn of digi and after that put phone into sleep (quick) 
2) than system power kernel event 507 can start digi again after screen is unlocked (this took around 15s, but hey it works)

Note its monkey kungfu workaround to be honest... As I could not find way how to start some action before sleep Its necessary tu turn off any action of power button + disable turning off the screen and doing point 1 just by script. If not done without turbing of digi only way is restart (so its good prevention) 

At least this I could findout after few painful hours... If anybody would like to help with this i can maybe provide more details.

But I would be really happy to have proper fix, or at least somehow improved this monkey kungfu eq with binding power key action to scrip. This is something what i am not really sure how its even possible that in year 2024 we cant do this in windows in some normal way as a feature :D