edk2-porting/edk2-msm #318: Support for Poco x5 PRO

RIllEX0, 2023-08-14:
Please add support for Poco X5 PRO 5G Redwood
8/256 (snapdragon 778G, 8 cores)
Please, i want to test windows 11 on fast and strong phone, but i cant because of no support/ no unofficial too
My discord (Type me in discord for any answers in this issue, please🤥): `nxori_`

nedokaka, 2023-08-14:
@RIllEX0 No.

RIllEX0, 2023-08-14:
why😟

nedokaka, 2023-08-14:
@RIllEX0 Because i said no

halal-beef, 2023-08-14:
@RIllEX0

RIllEX0, 2023-08-14:
yes? @halal-beef

nedokaka, 2023-08-14:
@halal-beef

nedokaka, 2023-08-14:
@nedokaka Xd

JSPiRiT1337, 2023-08-14:
soon, almost ready

RIllEX0, 2023-08-14:
> soon, almost ready

Really?!? Thank you so much, if this real

rxuglr, 2023-08-14:
This is fake

RIllEX0, 2023-08-14:
@JSPiRiT1337 How much soon? like a week? thanks for answering

halal-beef, 2023-08-14:
fake

rxuglr, 2023-08-14:
real

koke228, 2023-08-14:
halal beef nigger

halal-beef, 2023-08-14:
@koke228 LMAO

JSPiRiT1337, 2023-08-14:
@halal-beef lmao dude just wait a couple of days and you'll see

nedokaka, 2023-08-14:
@koke228

RIllEX0, 2023-08-14:
bruh, random russian Илья, stop being dumbass, i asked for SMART question  and you just spam in issue
go touch grass nig-

rxuglr, 2023-08-14:
@RIllEX0 Poco x5 pro

halal-beef, 2023-08-14:
@rxuglr one must imagine Sisyphus oiled up

RIllEX0, 2023-08-14:
@rxuglr yes, poco x5 pro, i guess this is strong phone, and windows gonna be flying on this phone without lags
but, who knows🤓

nedokaka, 2023-08-14:
@RIllEX0

rxuglr, 2023-08-14:
@RIllEX0 It cursed phone. Pharaohs curse

halal-beef, 2023-08-14:
@rxuglr Return the slab

rxuglr, 2023-08-14:
@halal-beef Who are you

halal-beef, 2023-08-14:
@rxuglr Good risings Pharaoh

RIllEX0, 2023-08-14:
One more tip to contributors and others, poco x5 pro already got unofficial, but stable twrp and ofox recovery (+ lot of custom roms)
soooo, just saying🤥

halal-beef, 2023-08-14:
but my TV has mainline Linux

rxuglr, 2023-08-14:
@RIllEX0 I'm gonna take away unofficial twrp and ofox from Poco x5 pro

RIllEX0, 2023-08-14:
wdym?

nedokaka, 2023-08-14:
@RIllEX0 Poco x3 pro:
395886 billion custom roms
Ofox,twrp,pbrp,shrp,fprb
Droidian, manjaro arm, sfos, ubt and windows
And broken mainline😁

halal-beef, 2023-08-14:
he's saying he going to make it official :trollface:

rxuglr, 2023-08-14:
@RIllEX0 You know what I mean☺️

RIllEX0, 2023-08-14:
@nedokaka i meant, poco x5 pro, is new phone, lot of people think about that theres small count of recoverys and roms, so just saying

rxuglr, 2023-08-14:
@RIllEX0  poco x5 pro = outdated phone

halal-beef, 2023-08-14:
@rxuglr Poco X5 pro = modified Poco X3 nfc

rxuglr, 2023-08-14:
@halal-beef poco = poor