edk2-porting/edk2-msm #323: I'm masturbating on Snapdragon 855

pidorkalobrod, 2023-08-25:
What i should do?

TheBACKTRACK, 2023-08-25:
@pidorkalobrod Maybe... You should stop?

SnowFuhrer, 2023-08-25:
@pidorkalobrod :|

TheBACKTRACK, 2023-08-25:
@SnowFuhrer Пидор калоброд

halal-beef, 2023-08-25:
so real!

TheBACKTRACK, 2023-08-25:
I think you should stop now

mrksbgg, 2023-08-25:
Dont stop

halal-beef, 2023-08-25:
forbidden thermal paste

mrksbgg, 2023-08-25:
> forbidden thermal paste

thermal paste for ryzen 9

TheBACKTRACK, 2023-08-25:
Best thermal paste for Intel Core i5 10300H

J0SH1X, 2023-08-25:
@mrksbgg never eat halal its bad

halal-beef, 2023-08-25:
> @mrksbgg never eat halal its bad

whatever u say bruder

J0SH1X, 2023-08-25:
@J0SH1X doner kebab is mostly from halal meet

mrksbgg, 2023-08-25:
> @mrksbgg never eat halal its bad

why halal is bad

J0SH1X, 2023-08-25:
@mrksbgg he will port edk2 to your brain

halal-beef, 2023-08-25:
> @J0SH1X doner kebab is mostly from halal meet

doner good men

halal-beef, 2023-08-25:
u have unlimited turkey people too so infinite doner

halal-beef, 2023-08-25:
> What i should do?

is it done

halal-beef, 2023-08-25:
or are u still going

nedokaka, 2023-08-25:
@pidorkalobrod ok

closed 2023-08-26 by gus33000