edk2-porting/edk2-msm #326: MediaTek Dimensity 8050 port UEFI and Windows please

pidorgandon69, 2023-08-26:
I want Windows port for Tecno Camon 20 Pro 5G. I really want to play GTA 5 on my phone. MediaTek Dimensity 8050 have more performance than, for example, Ryzen 9 5900X.
So i will get 1000+ FPS in GTA 5
I think it has very good support of Windows
And also, this CPU can use ARM and x86 because i can play Half Life 2 in Android
HL2 is x86 game, so, this CPU can switch to x86
I think that porting Windows to this CPU will be very easy because this CPU can switch to x86 from ARM 
I can help because i'm expert in Ubuntu, for example, not so many people know about Nano text editor. And also, switching to other desktop environment is very hard
But i don't know anything about Arch Linux because Arch is hardest Linux distro

nedokaka, 2023-08-26:
@pidorgandon69 idi nahui

TheBACKTRACK, 2023-08-26:
@pidorgandon69 Ты еблан?

govnoed1488dorogayayakupiltebehuinu911, 2023-08-26:
Почему я не могу просто взять и выйти на площадь голым и начать портировать винду на деменцию 8050. А после насиловать всех прохожих в анальное отверстие и обмазывать их фекалиями. После помочится на прохожих при этом повторяя "Хуй каки, каки париром пом, пом". Потом через минуту вернутся домой, взять с собой швабру и сметану, вернутся на площадь, раздеть прохожего и обмазать его анус сметаной после засунуть в него шваброй и начать его насиловать. Потом через полчаса засунуть швабру так глубоко, что она вылезет у него из рта, и когда он задохнётся вынуть её всю в фекалиях, поднять вверх и закричать: "Я викинг могучего хуя властилина!" и избивать насмерть всех прохожих. А, через полчаса справить нужду по большому в банку сметаны и зайти в ближайшее кафе с криками "Я джедай пизды негра! Пришёл сюда смазать очко сучкам!" и кидаться содержимым в банке в людей и их насиловать. После украсть в супермаркете масло, угнать чей-то велосипед и катаясь по площади обливать всех маслом крича "Вы все вафлёры, а я бог солнца" . Когда масло закончится, крикнуть "Сочные сиськи" и набросится на ближайшего человека и насиловать его до конца дня.

ETCHDEV, 2023-08-26:
What are you talking about, even I know about nano and more. No, it doesn't support x86 and half life 2 is ported for arm

muhammad23012009, 2023-08-26:
@pidorgandon69 nice bro, good luck

vladimirlishchuk, 2023-08-26:
lmao

closed 2023-08-26 by TAOcroatia