edk2-porting/edk2-msm #330: skill issue

govnoed1488dorogayayakupiltebehuinu911, 2023-08-26:
### Prerequisites

* [ ] Have you read the wiki? no, fuck you
* [ ] Is your device in the supported list? yes, redmi 6a
* [ ] Does the device have a Snapdragon SOC which is supported in this repo? yes, snapdragon mt6577

### Description

[Description of the bug or feature]
hey, gays. my name is ronnie mcnutt, i was born may 23, 1987 and was a resident of New Albany, Mississippi and had served in the United States Army Reserve, including in iraq. i worked at toyota plant. i've had a variety of mental health problems, such as depression as well as stress disorder, the latter of which was a direct consequence of my time served in the iraq war in 2007 and 2008

**Expected behavior:** [What you expected to happen] facebook stream

**Actual behavior:** [What actually happened] redmi 6a is dead

halal-beef, 2023-08-26:
@govnoed1488dorogayayakupiltebehuinu911 This one crazy

govnoed1488dorogayayakupiltebehuinu911, 2023-08-26:
> @govnoed1488dorogayayakupiltebehuinu911 This one crazy

im a ronnie mcnnot with redmi 6a

closed 2023-08-26 by TAOcroatia