edk2-porting/edk2-msm #362: How do you patch ButtonsDxe?

jonahnm, 2023-10-13:
### Prerequisites

* [Yes] Have you read the wiki?
* [No] Is your device in the supported list?
* [Yes] Does the device have a Snapdragon SOC which is supported in this repo?

### Description

[Description of the bug or feature]

**Expected behavior:** [What you expected to happen]
Detailed instructions in the documentation.
**Actual behavior:** [What actually happened]
No documentation on how to patch ButtonsDxe.

AistopGit, 2023-10-14:
https://github.com/edk2-porting/renegade-project.org/blob/master/en/sm8150pkg/tutorial/4_Preparing_files_workspace.md begin reading from here

jonahnm, 2023-10-19:
> https://github.com/edk2-porting/renegade-project.org/blob/master/en/sm8150pkg/tutorial/4_Preparing_files_workspace.md begin reading from here

Documentation should still be provided in a non-archived repository.