edk2-porting/edk2-msm #377: Xiaomi Mi A1

MuhammedEmirArtan, 2024-01-23:
### Prerequisites

* [ yes] Have you read the wiki?
* [no ] Is your device in the supported list?
* [qualcomm snapdragon 625 and i think yes] Does the device have a Snapdragon SOC which is supported in this repo?

### Description
so when you will add support for the mi a1?
[Description of the bug or feature]
i want to install win 10 / 11 
**Expected behavior:** [What you expected to happen]
seeing mi a1 on the supported devices list
**Actual behavior:** [What actually happened]
no

rxuglr, 2024-01-23:
@MuhammedEmirArtan What

MuhammedEmirArtan, 2024-01-23:
@rxuglr umm yes

NetuserPRO, 2024-01-23:
@MuhammedEmirArtan add it yourself

MuhammedEmirArtan, 2024-01-23:
@NetuserPRO do you mean porting

rxuglr, 2024-01-23:
@MuhammedEmirArtan Yup

NetuserPRO, 2024-01-23:
@MuhammedEmirArtan yes ofc, and windows on 625 doesn't have drivers

MuhammedEmirArtan, 2024-01-23:
so it will not work

NetuserPRO, 2024-01-23:
@MuhammedEmirArtan it CAN work, just don't expect much

MuhammedEmirArtan, 2024-01-23:
hmmm
so am i have to do anything additional with the normal installation or porting

MuhammedEmirArtan, 2024-01-23:
https://xdaforums.com/t/driver-adreno-5xx-gpu-driver-v313-rev23-v331-rev1.3815853/
@NetuserPRO

MuhammedEmirArtan, 2024-01-23:
oh wait

MuhammedEmirArtan, 2024-01-23:
this is an android driver :skull:

NetuserPRO, 2024-01-23:
@MuhammedEmirArtan porting, a lot of porting.

MuhammedEmirArtan, 2024-01-23:
hmm so i have to rewrite CPU driver 
can i do it with a template

NetuserPRO, 2024-01-23:
@MuhammedEmirArtan figure it out yourself

MuhammedEmirArtan, 2024-01-23:
nah i give up

closed 2024-01-23 by MuhammedEmirArtan