edk2-porting/edk2-msm #382: Build failed for Xiaomi Yudi (SM8475) device

FeatherXavier, 2024-04-06:
Command excuted: ./build.sh -d yudi

Verbose output:

> Ubuntu clang version 14.0.0-1ubuntu1.1
Target: aarch64-unknown-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
Found candidate GCC installation: /usr/bin/../lib/gcc-cross/aarch64-linux-gnu/11
Selected GCC installation: /usr/bin/../lib/gcc-cross/aarch64-linux-gnu/11
Candidate multilib: .;@m64
Selected multilib: .;@m64
 "/usr/bin/aarch64-linux-gnu-ld" -pie -EL -z relro --hash-style=gnu --build-id --eh-frame-hdr -m aarch64linux -dynamic-linker /lib/ld-linux-aarch64.so.1 -o /mnt/i/edk2-msm/workspace/Build/yudi/RELEASE_CLANG38/AARCH64/src/main/SimpleInitMain/DEBUG/SimpleInitMain.dll -L/mnt/i/edk2-msm/Common/edk2/ArmPkg/Library/GccLto -u _ModuleEntryPoint -L/usr/bin/../lib/gcc-cross/aarch64-linux-gnu/11 -L/usr/bin/../lib/gcc-cross/aarch64-linux-gnu/11/../../../../lib64 -L/lib/aarch64-linux-gnu -L/lib/../lib64 -L/usr/lib/aarch64-linux-gnu -L/usr/lib/../lib64 -L/usr/bin/../lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/lib -L/usr/lib/llvm-14/bin/../lib -L/lib -L/usr/lib -plugin /usr/lib/llvm-14/bin/../lib/LLVMgold.so -plugin-opt=mcpu=cortex-a53 -plugin-opt=O2 -plugin-opt=-function-sections -plugin-opt=-data-sections --emit-relocs --gc-sections -e _ModuleEntryPoint -Map /mnt/i/edk2-msm/workspace/Build/yudi/RELEASE_CLANG38/AARCH64/src/main/SimpleInitMain/DEBUG/SimpleInitMain.map -z common-page-size=0x20 -z common-page-size=0x1000 -O3 -llto-aarch64 -plugin-opt=-pass-through=-llto-aarch64 -z common-page-size=0x1000 --start-group @/mnt/i/edk2-msm/workspace/Build/yudi/RELEASE_CLANG38/AARCH64/src/main/SimpleInitMain/OUTPUT/static_library_files.lst --end-group --script=/mnt/i/edk2-msm/Common/edk2/BaseTools/Scripts/GccBase.lds --defsym=PECOFF_HEADER_SIZE=0x228
/usr/bin/aarch64-linux-gnu-ld: /tmp/lto-llvm-00c95f.o:(.data.rel.ro.ft_default_modules+0x80): undefined reference to `ft_sdf_renderer_class'
/usr/bin/aarch64-linux-gnu-ld: /tmp/lto-llvm-00c95f.o:(.data.rel.ro.ft_default_modules+0x88): undefined reference to `ft_bitmap_sdf_renderer_class'
/usr/bin/aarch64-linux-gnu-ld: /tmp/lto-llvm-00c95f.o:(.data.rel.ro.ft_default_modules+0x90): undefined reference to `ft_svg_renderer_class'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [GNUmakefile:459: /mnt/i/edk2-msm/workspace/Build/yudi/RELEASE_CLANG38/AARCH64/src/main/SimpleInitMain/DEBUG/SimpleInitMain.dll] Error 1

Building on WSL Ubuntu, gcc version 11.4.0, clang 14. It seems to be the problem of FreeType?

Ost268, 2024-04-06:
Try /build.sh -d yudi --fixclang

FeatherXavier, 2024-04-13:
Not working. Still the same error.

获取Outlook for Android<https://aka.ms/AAb9ysg>