edk2-porting/edk2-msm #400: Doubt to port for Nexus 6p, it will have the UFS cleared?

Bezouro, 2024-07-15:
### Prerequisites

* [x] Have you read the wiki?
* [ ] Is your device in the supported list?
* [ ] Does the device have a Snapdragon SOC which is supported in this repo?

### Description

I want to port to the Nexus 6p and I have noticed about the UFS cleaning on Google devices, so I have the doubt, should I consider Nexus devices as Google devices or as Huawei in the case of the 6p?

Will my angler have clear UFS?

gus33000, 2024-07-15:
That phone shouldn't have an ufs. It should be using an emmc. So this shouldn't apply at all.

Bezouro, 2024-07-15:
thanks @gus33000 
ahh, wonderful
as a hardware newbie I thought they would be the same thing lol

I also had doubts because the diagram does not make clear the implications of having the device wiped
[image]

J0SH1X, 2025-10-16:
Angler is supported here windows can boot dont ask about drivers tho 

https://github.com/Dominduchami/Nexus5XPkg

closed 2025-10-16 by J0SH1X