edk2-porting/edk2-msm #405: An explination of why patching buttonsdxe usbconfigdxe is needed?

bm16ton, 2024-09-09:
### Prerequisites

* [ ] Have you read the wiki?
Yes and has an excellent how to patch these binaries when a patched bin for model exists
* [ ] Is your device in the supported list?
 NOPE
* [ ] Does the device have a Snapdragon SOC which is supported in this repo?
Im not sure if the sm8550 has reached "supported", but code for sum of  it exists.
### Description

Hello, In general Im curious about the patching of buttonsdxe and usbconfigdxe. None of the info I can find mentions why these need to be patched in order to work, what function we are actually patching, and what information we are changing, and why what we are changing it to. The how to was great, unfortunately my OnePlus 12r uses sm8550 and I have not found a patched buttonsdxe for sm8550 (except for samsung zfold and its buttonsdxe differs greatly) And without this base knowledge trying to manually patch becomes considerably harder. Thank you guys, love the great work! edk2 is simply amazing.

TrustedFloppa, 2024-09-09:
sar, buttonsdxe are for navigating in uefi, and usbconfigdxe is for usb

J0SH1X, 2024-09-09:
@bm16ton buttonsdxe patches the char returned by powerkey to be enter

bm16ton, 2024-09-09:
@JOSH1X Thankyou! That is one down and currently the most important one. Tho with none of the buttons working it seems this patch issue is not the only one im facing. Plus Oneplus (all sm8550?) took volume down from regular gpio and added it to pmics along with power. And just a sidenote for anyone stumbling here, the device tree entries for Volume up list the keycode for volume down, and the the volume down (no longer in gpio keys section) lists the key for volume up. Despite this they both work correctly (vol+ = vol+ etc) so not sure where its being switched, food for thought.

Govro150, 2024-09-10:
That device is ported in Silicium so you can take ( or repatch ) dxe from other project