edk2-porting/edk2-msm #411: support for SM6150

nookgit12, 2025-01-19:
plz i dont know how to write these config files so i want to ask yall to make config for SM6150 (snapdragon 675)

sndnvaps, 2025-04-13:
@nookgit12 i have init the project for xiaomi note 7 pro(violet), platform is sm6150
now it can build boot-violet.img, but need more time to dump the dsdt.aml from my cell phone

sndnvaps, 2025-05-08:
when i try to port edk2 to sm6150 platform, i had got some trouble
like this

[image]

is any one can give me some advices....

Sasmithere, 2025-09-20:
can you send the boot-violet.img ? i used dipper img(personal modified version) to use split emmc partitions to win esp etc ...
so far i am not able to make a boot img for it

sndnvaps, 2025-09-21:
@Sasmithere in xiaomi_violet-edk2-boot-violet.zip contains 2 file, edk2-boot-violet.img and edk2-boot-violet-old.img

[xiaomi_violet-edk2-boot-violet.zip](https://github.com/user-attachments/files/22451810/xiaomi_violet-edk2-boot-violet.zip)

Sasmithere, 2025-09-21:
i'll try both and return with an update

Sasmithere, 2025-09-22:
@sndnvaps the old img one seems to boot into efi shell but my otg usb keyboard doesn't seem to work , i was wondering if you still have the .dsc , .fdf and .conf related files for it , i want to change the cmdline so it automatically boots to windows partition

Thirumal-Palwai, 2025-10-01:
Hey @sndnvaps @Sasmithere 

I came across this project and thought it might be useful for your work:

GitHub: https://github.com/Project-Aloha/mu_aloha_platforms
Porting Guide: https://aloha.firmware.icu/PortingGuides/SimpleGuide.html

Sasmithere, 2025-10-01:
@sndnvaps thanks for the files ,
 The graphicsconsoledxe.efi is causing the ruckus I'll be approaching this issue in these ways :
1. fix the graphicsconsoledxe.efi by dumping and recompiling then insert some dummy registers which are causing the issue and i did saw some registers having <unknown> as address (hard)
2. use a dummy graphicsconsoledxe.efi which returns efi_success as true and goes to text based interface (medium)
3. change the stack in which graphicsconsoledxe is called and remove it from there , inshort remove it from the call stack(hard) 
4. build graphicsconsoledxe from scratch myself from the xbl.elf file or search internet for something similar (very hard 💀)
5. and last is i will try to change the bootdef.c and boot.c so it automatically boots into uefi boot menu

Sasmithere, 2025-10-02:
@sndnvaps 

[image]

is emmc storage not supported ?

Sasmithere, 2025-10-02:
close 🤏🏻  but no touchscreen and pwm , vol_up and down work  , just scripted to auto boot to simple-init 

[image]

sndnvaps, 2025-10-29:
@Sasmithere i cherry-pick you changed, but still get dark screen. you need to push patched file to https://github.com/Sasmithere/edk2-msm-binary/tree/main/Drivers/sm6150  , so i can make a fully test

Sasmithere, 2025-10-29:
@sndnvaps  i'll upload here my current working set

[edk2-violet.zip](https://github.com/user-attachments/files/23201000/edk2-violet.zip)

sndnvaps, 2025-10-29:
@Sasmithere in this commit https://github.com/BigfootACA/simple-init/blob/29e062805776db5ab7a61d07ef040d43aed71161/libs/lvgl/README.md, Features, support touch screen, 
it make do some hack to make it work....

Sasmithere, 2025-10-31:
@sndnvaps TouchScreen config is in DisplayDxe.efi , but when i include it , it restarts the device

Sasmithere, 2025-10-31:
i made WLED, IBB, LAB, LCDB to return EFI_SUCCESS and it work, but no touch still but before booting to tianocore  ,debug error : 

```
MDPSystem: Failed to set variable <DisplaySupportedPanelCount>. Error: 00000002
MDPSystem: Failed to set variable <DisplaySupportedPanelList>. Error: 00000002

```
0x2 likely meaning not found

[StubPmicDxe.c](https://github.com/user-attachments/files/23268700/StubPmicDxe.c)

corresponding inf :
 ```
[Defines]
  INF_VERSION    = 0x00010005
  BASE_NAME      = StubPmicDxe
  FILE_GUID      = 12345678-1234-1234-1234-123456789ABC
  MODULE_TYPE    = DXE_DRIVER
  VERSION_STRING = 1.0
  ENTRY_POINT    = StubPmicEntry

[Sources]
  StubPmicDxe.c

[Packages]
  MdePkg/MdePkg.dec

[LibraryClasses]
  UefiDriverEntryPoint
  UefiBootServicesTableLib

[Depex]
  TRUE
```

J0SH1X, 2025-11-01:
Touch in uefi needs a driver aka touchscreendxe but there is only one for synaptics Touchscreens and also iirc to make touch work you need to recompile i2cdxe and gpidxe (at least on sm8150)

sndnvaps, 2025-11-01:
@J0SH1X you mean do some hack to I2C.efi & GpiDxe.efi to make the TouchScreen work.

Sasmithere, 2025-11-01:
@sndnvaps i patched the .efi and added Panel_shenchao_nt36672a_fhd_video.xml from raw files no reboots now and ,i follwed sm8150 pattern DisplayDxe.efi @J0SH1X , but still no touch .... need to hack i2cdxe and gpidxe ?, what to change in i2cdxe and gpidxe exactly ?i have no idea on that one

[DisplayDxe.zip](https://github.com/user-attachments/files/23286456/DisplayDxe.zip)

sndnvaps, 2025-11-02:
@Sasmithere i have not idea too. 
@J0SH1X i want to compile the I2cdxe from source, but cannt build in src tree.  can give us some idea, how to hack the i2cdxe & gpidxe

Sasmithere, 2025-11-04:
@sndnvaps 
my current implementation shows that usb is detected in tianocore , will that help ? i have attached below
debug
```
UsbConfigPortsQueryConnectionChange: Detach and Type-C

UsbConfigPortsQueryConnectionChange: usbport->connectstate: DEATT

ConfigPortsQueryConnectionChange: UFP and Type-C 

UsbConfigPortsQueryConnectionChange: usbport->connectstate: ATT
```
[Apriori.Dxe.zip](https://github.com/user-attachments/files/23338039/Apriori.Dxe.zip)

sndnvaps, 2025-11-05:
> [@sndnvaps](https://github.com/sndnvaps) my current implementation shows that usb is detected in tianocore , will that help ? i have attached below debug
> 
> ```
> UsbConfigPortsQueryConnectionChange: Detach and Type-C
> 
> UsbConfigPortsQueryConnectionChange: usbport->connectstate: DEATT
> 
> ConfigPortsQueryConnectionChange: UFP and Type-C 
> 
> UsbConfigPortsQueryConnectionChange: usbport->connectstate: ATT
> ```
> 
> [Apriori.Dxe.zip](https://github.com/user-attachments/files/23338039/Apriori.Dxe.zip)
> 
> i got this debug below in loggerd viewer
> 
> ```
> lvgl: [Error] (180.479, +180479) _lv_inv_area: detected modifying dirty areas in render (in iv refr.c line #212)
> lvgl: [Error] (180.479, +0) _lv_inv_area: detected modifying dirty areas in render (in iv refr.c line #212)
> lvgl: [Error] (180.479, +0) _lv_inv_area: detected modifying dirty areas in render (in iv refr.c line #212)
> lvgl: [Error] (180.479, +0) _lv_inv_area: detected modifying dirty areas in render (in iv refr.c line #212)
> lvgl: [Error] (180.479, +0) _lv_inv_area: detected modifying dirty areas in render (in iv refr.c line #212)
> lvgl: [Error] (180.479, +0) _lv_inv_area: detected modifying dirty areas in render (in iv refr.c line #212)
> lvgl: [Error](180.480, +1) _lv_inv_area: detected modifying dirty areas in render (in iv refr.c line #212)
> lvgl: [Error] (180.480, +0) _lv_inv_area: detected modifying dirty areas in render (in iv refr.c line #212)
> lvgl: [Error] (180.480, +0) _lv_inv_area: detected modifying dirty areas in render (in iv refr.c line #212)
> lvgl: [Error] (180.480, +0) _lv_inv_area: detected modifying dirty areas in render (in iv refr.c line #212)
> lvgl: [Error] (180.480, +0) _lv_inv_area: detected modifying dirty areas in render (in iv refr.c line #212)
> ```

@Sasmithere i had got the same log of lvgl

```
lvgl: [Error] (180.480, +0) _lv_inv_area: detected modifying dirty areas in render (in iv refr.c line #212 
```
and what you changed to make the usb work

i have buy a usb-a-type-c ttl cable, i hope the usb uart can work

sndnvaps, 2025-11-05:
@Sasmithere i will do the test this night. 
i will change the  
```
gQcomTokenSpaceGuid.PcdDebugUartPortBase|0xa88000
```
to 
```
 gQcomTokenSpaceGuid.PcdDebugUartPortBase|0x880000
```
uart port defind in boot.img file, cmd_line
```
erlycon=msm_geni_serial,0x880000 
```

sndnvaps, 2025-11-16:
@Sasmithere it seem not thing changed....
touch_screen still not work. 

but int loggerd viewer(we still got the same log)
```
guidrv: try init input driver uefi-touch
uefitouch: found new uefi absolute 0xc9b551f0
uefitouch: receive touch device hot-plug event
```
it may be fix the touch_screen will be easy then fix sdccDxe

sndnvaps, 2025-12-06:
@Sasmithere i will test you code today.
and respond the i2c_bus & i2c_addr

sndnvaps, 2025-12-06:
@Sasmithere use your driver. the gui res: 1080x2160, 
but not got the i2c error, it seem works. but TouchScreen still OffLine.

at the last, i got the execption occurred

[image]

J0SH1X, 2025-12-07:
> @sndnvaps <https://github.com/sndnvaps> i found something critical , i2c
> bus 4 is for display panel but not for touch controller , the touch
> controller should be on i2c bus 1 at 0x62 but uefi does not open for touch
> controller , it only opens for display panel through which we see the boot
> up

I told you you need to patch and recompile dalsys and gpi cause of that
exact iusse

Sasmithere, 2025-12-25:
@sndnvaps 
[patches.zip](https://github.com/user-attachments/files/24336145/patches.zip)
1. DALSYSDxe patched to enable i2c 1
2. PILDxe patched to cause any fails due to "Auto processing of uefipil.cfg failed. Calling target panic"
3. DisplayDxe patched from Lavender (redmi note 7) to fallback to shenchao xml instead of VirtioDSI fallback

@J0SH1X  can you tell me more about how can i enable or wake up the touch controller? manual gpio waking config crashes the device

sndnvaps, 2026-01-01:
@Sasmithere You need to upload the NovatekNvtTsDxe.depex & NovatekNvtTsDeviceDxe.depex for the test . only with *dxe.efi cann't work. we need the depex file to deal with the depend link.

sndnvaps, 2026-01-07:
@Sasmithere i have test it . touch screen not work for me.

Sasmithere, 2026-01-07:
@sndnvaps send your adb dump for NVT-ts or focal tech or himax idk, maybe your panel is different

sndnvaps, 2026-01-07:
i got this from the Recovery( which enable touch screen)
···
violet:/ # dmesg | grep "NVT-ts"
dmesg | grep "NVT-ts"
[    0.794297] [NVT-ts] nvt_driver_init 2321: start
[    0.794330] [NVT-ts] nvt_driver_init 2328: LCM is right! [Vendor]shenchao [IC] nt36672a
[    0.808765] [NVT-ts] nvt_ts_probe 1546: start
[    0.808807] [NVT-ts] nvt_parse_dt 785: novatek,irq-gpio=89
[    0.893837] [NVT-ts] nvt_ts_check_chip_ver_trim 1184: buf[1]=0x0A, buf[2]=0x00, buf[3]=0x00, buf[4]=0x72, buf[5]=0x66, buf[6]=0x03
[    0.893898] [NVT-ts] nvt_ts_check_chip_ver_trim 1210: This is NVT touch IC
[    0.948077] [NVT-ts] nvt_read_pid 431: PID=5916
[    0.948404] [NVT-ts] nvt_ts_probe 1704: int_trigger_type=1
[    0.948514] [NVT-ts] nvt_ts_probe 1716: request irq 182 succeed
[    0.948970] [NVT-ts] nvt_get_oem_data 608: ++
[    0.993557] [NVT-ts] Init_BootLoader 346: Init OK
[    1.036696] [NVT-ts] Resume_PD 189: Resume PD OK
[    1.140322] [NVT-ts] nvt_get_oem_data 706: --
[    1.140359] [NVT-ts] nvt_get_xiaomi_lockdown_info 723: The maker of Touch Panel & CG Lamination: 0x41
[    1.140418] [NVT-ts] nvt_get_xiaomi_lockdown_info 725: Display maker: 0x37
[    1.140477] [NVT-ts] nvt_get_xiaomi_lockdown_info 727: CG ink color: 0x32
[    1.140510] [NVT-ts] nvt_get_xiaomi_lockdown_info 729: HW version: 0x01
[    1.140569] [NVT-ts] nvt_get_xiaomi_lockdown_info 731: Project ID: 0x00F7
[    1.140603] [NVT-ts] nvt_get_xiaomi_lockdown_info 733: CG maker: 0x31
[    1.140661] [NVT-ts] nvt_get_xiaomi_lockdown_info 735: Reservation byte: 0x00
[    1.140720] [NVT-ts] lct_nvt_tp_info_node_init (line 751): LCM information : nt36672a video mode dsi shenchao panel
[    1.140958] [NVT-ts] nvt_flash_proc_init 647: Succeeded!
[    1.141017] [NVT-ts] nvt_flash_proc_init 650: ============================================================
[    1.141049] [NVT-ts] nvt_flash_proc_init 651: Create /proc/NVTflash
[    1.141108] [NVT-ts] nvt_flash_proc_init 652: ============================================================
[    1.141169] [NVT-ts] nvt_extra_proc_init 792: create proc/nvt_fw_version Succeeded!
[    1.141205] [NVT-ts] nvt_extra_proc_init 800: create proc/nvt_baseline Succeeded!
[    1.141265] [NVT-ts] nvt_extra_proc_init 808: create proc/nvt_raw Succeeded!
[    1.141326] [NVT-ts] nvt_extra_proc_init 816: create proc/nvt_diff Succeeded!
[    1.141445] [NVT-ts] nvt_mp_proc_init 2069: create /proc/nvt_selftest Succeeded!
[    1.141511] [NVT-ts] nvt_init_touchmode_data 1303: ENTER
[    1.141572] [NVT-ts] nvt_init_touchmode_data 1353: mode:0, set cur:0, get cur:0, def:0 min:0 max:1
[    1.141631] [NVT-ts] nvt_init_touchmode_data 1353: mode:1, set cur:0, get cur:0, def:1 min:0 max:1
[    1.141666] [NVT-ts] nvt_init_touchmode_data 1353: mode:2, set cur:0, get cur:0, def:0 min:35 max:50
[    1.141727] [NVT-ts] nvt_init_touchmode_data 1353: mode:3, set cur:0, get cur:0, def:112 min:64 max:255
[    1.141786] [NVT-ts] nvt_init_touchmode_data 1353: mode:4, set cur:0, get cur:0, def:0 min:0 max:0
[    1.141846] [NVT-ts] nvt_init_touchmode_data 1353: mode:5, set cur:0, get cur:0, def:0 min:0 max:0
[    1.141905] [NVT-ts] nvt_init_touchmode_data 1353: mode:6, set cur:0, get cur:0, def:0 min:0 max:0
[    1.141939] [NVT-ts] nvt_init_touchmode_data 1353: mode:7, set cur:2, get cur:2, def:2 min:0 max:3
[    1.141999] [NVT-ts] nvt_init_touchmode_data 1353: mode:8, set cur:0, get cur:0, def:0 min:0 max:3
[    1.142058] [NVT-ts] nvt_init_touchmode_data 1353: mode:9, set cur:0, get cur:0, def:0 min:0 max:0
[    1.142117] [NVT-ts] nvt_init_touchmode_data 1353: mode:10, set cur:0, get cur:0, def:0 min:0 max:0
[    1.142177] [NVT-ts] nvt_init_touchmode_data 1353: mode:11, set cur:0, get cur:0, def:0 min:0 max:0
[    1.142211] [NVT-ts] nvt_init_touchmode_data 1353: mode:12, set cur:0, get cur:0, def:0 min:0 max:0
[    1.142270] [NVT-ts] nvt_init_touchmode_data 1353: mode:13, set cur:0, get cur:0, def:0 min:0 max:0
[    1.142329] [NVT-ts] nvt_init_touchmode_data 1353: mode:14, set cur:0, get cur:0, def:0 min:0 max:0
[    1.142388] [NVT-ts] nvt_init_touchmode_data 1353: mode:15, set cur:0, get cur:0, def:0 min:0 max:0
[    1.142446] [NVT-ts] nvt_init_touchmode_data 1353: mode:16, set cur:0, get cur:0, def:0 min:0 max:0
[    1.142479] [NVT-ts] nvt_init_touchmode_data 1353: mode:17, set cur:0, get cur:0, def:0 min:0 max:0
[    1.142538] [NVT-ts] nvt_init_touchmode_data 1353: mode:18, set cur:0, get cur:0, def:0 min:0 max:0
[    1.142596] [NVT-ts] nvt_init_touchmode_data 1353: mode:19, set cur:0, get cur:0, def:0 min:0 max:0
[    1.142690] [NVT-ts] nvt_ts_probe 1838: end
[   14.986719] [NVT-ts] Boot_Update_Firmware (line 1040): g_lcd_id = nt36672a video mode dsi shenchao panel
[   14.986773] [NVT-ts] Boot_Update_Firmware 1044: firmware version is shenchao.
[   14.986800] [NVT-ts] update_firmware_request 55: filename is novatek/shenchao_nt36672a_miui_f7b.bin
[   15.026707] [NVT-ts] Resume_PD 189: Resume PD OK
[   15.193946] [NVT-ts] Check_CheckSum 293: firmware checksum match
[   15.220259] [NVT-ts] Init_BootLoader 346: Init OK
[   15.263363] [NVT-ts] Resume_PD 189: Resume PD OK
[   15.303302] [NVT-ts] nvt_ts_resume 2001: Touch is already resume
[   15.317012] [NVT-ts] nvt_check_flash_end_flag 1015: nvt_end_flag=NVT (4E 56 54)
```

Sasmithere, 2026-01-08:
@sndnvaps your logs were a very crucial key !!  , now touch is verify fast and responsive ,i have added a live log to check if its working or not , previously it was a hit and miss for me as well but it worked so i did not think much of it , i have pushed the changed to github at [NovatekTouchDxe](https://github.com/Sasmithere/novatek-touch-dxe )

https://github.com/user-attachments/assets/a0a804e8-c6e9-45a8-a84f-622b33ea0192

[boot-violet.zip](https://github.com/user-attachments/files/24501380/boot-violet.zip)

Sasmithere, 2026-01-09:
@sndnvaps i'll disable EFI_DEVICE_ERROR for this and update the build for testing

Sasmithere, 2026-01-09:
@sndnvaps try this ,this 100% should work ,i have disabled many filters ,those filters were for broken firmware ,i have applied actual linux ones instead of my broken one , and this is a test that my build should work in original screens as well , now some ghost touches will be there . give feedback and logs 

[boot-violet.zip](https://github.com/user-attachments/files/24531112/boot-violet.zip)

sndnvaps, 2026-01-11:
all right, i need to resync the project from you change. can you push the change to github.com