edk2-porting/edk2-msm #413: make port for Poco x6 5G [Snapdragon 7s gen2]

ItsWinterBush, 2025-02-05:
we want windows  11 arm on poco x6 5g

gus33000, 2025-02-05:
Who is we?

ItsWinterBush, 2025-02-05:
i just want windows 11 arm on my phone
poco x6 5G

BigfootACA, 2025-02-05:
You can port edk2-msm to your device by yourself.

Example commits:
https://github.com/edk2-porting/edk2-msm/commit/7352c1a46b208ab90184cbc7e97b63d3088dc574
https://github.com/edk2-porting/edk2-msm/commit/7cac7889f06b721d3a02dfc1a24d76ba738cccbd
https://github.com/edk2-porting/edk2-msm/commit/982284b9937d649b216ca6f890e8b9002b0eee3b

ItsWinterBush, 2025-02-05:
> You can port edk2-msm to your device by yourself.
> 
> Example commits: [7352c1a](https://github.com/edk2-porting/edk2-msm/commit/7352c1a46b208ab90184cbc7e97b63d3088dc574) [7cac788](https://github.com/edk2-porting/edk2-msm/commit/7cac7889f06b721d3a02dfc1a24d76ba738cccbd) [982284b](https://github.com/edk2-porting/edk2-msm/commit/982284b9937d649b216ca6f890e8b9002b0eee3b)

thx bro

ItsWinterBush, 2025-02-05:
but how

meeo11, 2025-02-07:
stop creating issue with porting, port yourself. read wiki.

Genzmous, 2025-02-15:
please make a uefi for the poco x6 because i really need it for my work please

meeo11, 2025-02-26:
for your work? buy a pc.